esa/asn1scc #30: Porting parser and generator to produce Java

alexvoronov, 2015-05-17:
Hi,

It's not an "issue", it's a question, but I hope Issues is the right place to ask it.

I wrote an open-source [UPER runtime](https://github.com/alexvoronov/geonetworking/tree/master/asn1-uper) for Java, to encode/decode classes annotated with [asn1-datatypes](https://github.com/alexvoronov/geonetworking/tree/master/asn1-datatypes). But I don't have a compiler yet, so I convert `*.asn` files to annotated Java classes by hand. Now I'm looking into writing a compiler. Do you think I can reuse your parser and generator to produce Java? Do you have any tips where could I start? I see you have four different `Backend*` folders, I wonder if you have any developer notes that tell what is what?

Best Regards,
Alex

maxime-esa, 2015-05-17:
I am out of the office until 18/05/2015.

Note: This is an automated response to your message  "[asn1scc] Porting
parser and generator to produce Java (#30)" sent on 17/05/2015 14:57:56.

This is the only notification you will receive while this person is away.
This message and any attachments are intended for the use of the addressee or addressees only.
The unauthorised disclosure, use, dissemination or copying (either in whole or in part) of its
content is not permitted.
If you received this message in error, please notify the sender and delete it from your system.
Emails can be altered and their integrity cannot be guaranteed by the sender.

Please consider the environment before printing this email.

maxime-esa, 2015-05-18:
Dear Alex,
I think you can easily reuse the parser and backends to generate Java code. The backends use template files that you can easily adapt to another language. To start with, you will not need to change anything in the code just edit existing templates and invoke the tool with and invoke the tool as is.

First step:
-----------
edit "ch.stg" : this file contains the template for the C header file (containing type definitions in C). For each production, write the data types in Java in place of C. You will quickly get familiar with the StringTemplate syntax - check http://www.stringtemplate.org/ for help

Second step:
---------------
edit c_source.stg, it contains the template for initialize/check/uPER encoding and other functions (.c file)

Then try your changes calling asn1.exe -c -uPER ...

alexvoronov, 2015-05-18:
Thanks a lot! Stringtemplate.org and `ch.stg` are excellent pointers for me to start!
I'll post an update when I have something working :-)

closed 2015-05-18 by alexvoronov

maxime-esa, 2015-05-19:
Great. Thanks.