esa/asn1scc #57: Implement the OER encoding

cmr, 2016-04-28:
The [OER encoding](http://www.oss.com/asn1/resources/books-whitepapers-pubs/Overview%20of%20OER.pdf) is intended to be very easy to decode. It may be a good fit for this compiler.

maxime-esa, 2017-07-21:
What is the added value compared to uPER ?

eternaleye, 2017-07-21:
@maxime-esa:

- Byte-alignment is likely to generate smaller/faster code
- Canonical OER is of interest for cryptographic purposes, as ambiguity is deeply problematic there (my interest)
- There exist protocols that mandate OER (NEMA  STMP, Interledger)

maxime-esa, 2017-07-22:
@eternaleye 
Thanks for the explanation.
We have no use of that encoding in the space industry at the moment, but your point is taken.
Feel free to contact me privately if you need it in a commercial context, and I'll put you in contact with the company that can implement it (the estimate is around 4 months of work to get both C and Spark/Ada runtimes).

closed 2017-07-22 by maxime-esa