esa/asn1scc #67: Modular runtime - only enable used features

vgheo, 2016-09-07:
In the context of safety-critical applications dead code is not allowed in the release binary.
Moreover, floating point is forbidden by certain standards.

The request is to modularize the asn1scc runtime so that an application can only include the actually used parts (features).
Candidate features
- floating point
- encoding types (only include runtime for the required encodings..)
  - acn
  - PER 
  - BER
...

ttsiodras, 2016-09-08:
I know it's not what you are asking for, but keep in mind that a large part of what you want in terms of dead code can be "emulated" with...

- Compiling with `-fdata-sections -ffunction-sections`
- Linking with `-Wl,-gc-sections`

This will basically ask GCC's linker to "garbage collect" any unused functions and remove them from the final binary.

Ditto for floating point - if you compile with `-msoft-float` and use the linker to garbage collect, I believe all the RTL functions implementing math functionality will be gone.

Anyway, I will keep your request open here regardless - enhancement requests are useful in terms of guiding the compiler's future.

vgheo, 2016-09-24:
I've done a prototype of the proposed modularization :
https://github.com/ttsiodras/asn1scc/compare/master...vgheo:modular-crt?expand=1

It only builds with CDT/linux.
I have no idea how to build on other tools/platforms.

Please review and suggest enhancements required for merging upstream.

ttsiodras, 2016-09-26:
Hi vgheo.

First of all, thanks for doing the work you did - most people stop at "suggesting improvements".

However, if you want this work to be merged upstream, there are a number of cleanups that must be performed.

1. We target lean, safety-critical embedded systems - and that's usually the domain of Makefiles and automated build systems, not Eclipse or any of the other IDEs. I can appreciate the convenience of IDEs (I begun my career 20 years ago living and breathing in them) but I can't accept anything IDE-related in contributions to ASN1SCC's RTL. There's lot's of CDT-related files in there, which simply have no place in ASN1SCC.

2. It will be far easier to review your work piece-meal - if you create a huge pull request, it is very difficult for me or George to allocate the time and effort needed to review what you are doing... and decide on whether it has merit or not. You'll need to break down the changes into bite-sized pull requests, with small - but clear - standalone changes.

Again, thanks for doing this - it will be nice if we can merge your work and appeal to even more safety-critical audiences; but you have to help us by making it feasible as a task (both me and George are doing ASN1SCC and TASTE work on the side - in our free time - and can't review huge pull requests).

Hope you are OK with this - and regardless, thanks for putting in the work to implement your dead-code elimination. At the very least, your repo is there for those that care about it.

Cheers,
Thanassis.

vgheo, 2016-09-28:
Dear Thanassis,
Thank you for your appreciation. It's not that selfless, as I have a project at my job that involves asn1scc.
(current work still done in my own time...)

1. I'm in safety critical sw development since 10 years. We do have makefiles, but also use eclipse cdt on top for convenience.
The cdt-related files are 
.project - base eclipse project configutation 
.cproject - cdt specific project config
.settings -  subfolder with settings oef various cdt plugins
Cdt is nice and cross platform . I can't imagine how could i do this kind of refactoring without an ide, and on linux cdt is the only option i know of.
I disagree when you say " I can't accept anything IDE-related in contributions to ASN1SCC's RTL", as you already have visualstudio project files.
A downside is that if we go with makefile, cdt and vs, there are 3 buildsystems to maintain.
Please make a decision and tell me how to proceed. ( if you say only makefiles, vcproj files should not be maintained)
2. The main commit is a pure refactoring . I have only cut/pasted initial code in several files, to achieve modularity.
There were also some wierd external declarations in c files that i have removed, as they are now coming from the included headers.
I also see some extra whitespace that I've changed - I want to rework that so that no change is done on the original lines.
In order to review this you must use git blame -C on each file.
The standard diff does not help much.
3. I now realize that this change is also dependent on how the rtl is deployed in the user project.
The current way is that asn1scc copies the files in the output dir.
In my job project we have several protocols, each in its own source dir.
Therefore, we have isolated the rtl in a separate library project, so that it can be shared by the many protocols.
The idea would be to deliver the rtl as a library, not embedded with user protocol code.
It may even be a separate repo.
In any case, the file structure in the proposed change is made based on the assumption that rtl is a library where the user will need to include the headers, build the library and link the result in the final app.

Kind regards 
Vlad Gheorghe

ttsiodras, 2016-09-28:
Hi Vlad - good to hear you're using ASN1SCC at work :-)

Regarding Eclipse project files: I understand what you are saying - but I stand by what I said: a patch request with 22 files is a much different thing (reviewing effort-wise) to a patchset on 1 or 2 files. Either me or George can allocate some time to check the latter, where as the former is something different. As for "asn1crt.vcproj", that's something I personally never use. If the RTL is ever modified, a simple `gcc -c -Wall *.c` is more than enough to see if anything broke, IMHO. George may disagree though :-) 

Your idea about creating a library for the RTL is something that we prefer to let end-users decide for themselves. For many of the commercial ASN.1 compilers, the RTL libraries are black boxes, impossible to modfiy - ASN1SCC distinguishes itself in this as well, as the RTL is an open thing to see/hack/modify in your project as you see fit. If as you have described, you have multiple protocols, by all means do as you propose - packing the RTL in a lib for all of them. I'd do the same in your shoes - or have my Makefile only use the RTL set from one.

Cheers,
Thanassis.

vgheo, 2016-09-29:
Dear Thanassis,

I'm not sure I'm quite getting what you are suggesting.
If you agree, I'd continue this discussion split on specific topics.

** RTL modularsation **
Main feature.
Introduce proper modules (header/source code file pairs). 
   Existing code is grouped into cohesive modules.

**"cdt project files" - whether the CDT-specific files can be accepted into asn1scc**
In order to edit the RTL in CDT/Eclipse, any user needs to add these files.
The configuration is NOT trivial. Some configuration needs to be made manually.
CDT generates the makefiles on linux.

( The agile blurb :) 
As an RTL developer using CDT, 
I want these files maintained upstream
So that I don't have to re-do this setup every time I start hacking/contributing to this project. Otherwise I'll be terribly put off.

**"RTL filetree restructure" - in the proposed change, RTL code is split into separate include and src directories**
The intention is that the user code has
`# include "asn1scc/MODULE.h"` - include a single module
`#include "asn1scc.h" ` - full library
`-I <PATH_TO_ASN1SCC_RTL>/include` - gcc flag

Also, a makefile must be provided for a recursive build of the src/ directory.
- I can add this if agreed

I prefer this approach as it is modular with respect to an user application.
It also has the standard file structure of a library. (L in RTL).
Also - the user can choose whether to include this in his source tree (source deployment), or build a separate library and only provide the headers + library to the application build (binary deploment).
The decision is still with the user.
 
However, this decision must be consistent with "RTL deployment" - see next section.

**"RTL deployment" - how to deploy RTL to user projects**
Currently RTL deployment is done 
- in the SAME output directory as the user code
- EACH time the user generation is done
-- even though the RTL does not change that often

Issues
- violation of loose coupling/high cohesion
  - In a user application, RTL is a lower layer than the user-generated protocol code

Proposals
- ALT1: asn1scc separate option to generate RTL ONLY
-- this can be used in a user makefile as a separate target, with a different output directory
- ALT2: deliver the RTL as a separate source repository.
-- a user would clone this RTL repo, and add it as a dependency to the user's project

**Optional Real Feature**
- floating point sections are only available if ASN1SCC_REAL is defined
Is that OK ?
Then, it should be extended to other features like :
- BER
- XER

This would allow the user to select only the features that it uses. (talking lean embedded...)

**"patchset granularity" - whether the current proposal can be accepted as-is, or split into smaller chunks for easier review / merge**

The current change set is formed as follows:
ea3a :  "RTL modularisation"
  "CDT project files"
   NO CODE IS CHANGED.
  In order to review this please use `git blame -C FILE` on each path.
  Some whitespace is also changed, but I want to rework this so absolutely all lines are originating in pre-existing commits.

31b3: implement "RTL filetree restructure"
1fba: prototype for "Optional Real Feature"

9669 - "CDT config files" - fix
c999 - " Optional Real Feature" - fix
6f74 - "RTL Modularisaion" - fix
d4e0 - "RTL Modularisation" - cosmetic fix

My intention with this branch is to submit the code for review, get feed-back and then prepare the final pull requests. (this is why I did not make a pull request from the first place).
So I fully agree with you comment - this is not a cohesive, mergeable pull request.

Now, considering the acceptance of each feature, I will prepare corresponding pull requests.

Best regards
Vlad

ttsiodras, 2016-09-30:
Hi Vlad - prepare the merge requests and we'll discuss them individually.

ttsiodras, 2016-09-30:
Quick feedback (still at work): 

(1) one pull request for whitespace cleanup (Microsoft's sins - redefining things like TABs and newlines (sigh) - will definitely accept this one, to make the files appear cleanly in all OSes
(2) CDT configuration - here's a thought: it would be very nice if we had folders underneath the RTL with the build projects for each target (one for CDT, one for MS, one with a Makefile). Would merge such a thing, too.
(3) Optional REALs, BER, XER, etc - first of all, the latest version of the compiler doesn't emit BER and XER files in the output. Then again, even if it did, why does the -Wgc-sections suggestion doesn't cover you in terms of dead code (see my first reply in this discussion)? Its the object code that counts (even in category A software, like the one running inside the launchers we use in the European Space Agency).

Cheers,
Thanassis.

ttsiodras, 2016-09-30:
And one more:

(4) Maybe we can think about a "-noRTL" flag to avoid generating the files in use cases like yours (with multiple protocols in different folders).

vgheo, 2016-10-01:
But what about the "RTL modularization" feature ?
The idea is to have clean modules in the source code (as I understand the design intention) : 
- core.h - fundamental definitions
- util.{h,c} - basic utilities for other 
- real.{h,c}
- BitStream.{h,c}, ByteStream.{h,c}  - base abstractions
  - use core, util
- encodings
  - xer.{h,c}
  - ber.{h,c}
  - acn.{h,c}

In the current state on master 
- declarations are all mixed in asn1crt.h
- implementations are some in separate modules (eg. Acn.c) some mixed in asn1crt.c
This is 
- non-cohesive
- unneccessary coupled
- inconsistent 

It is not a big change, but I think it significantly enahnces the maintainability and understandability of the RTL.
Additionally, as originally stated, it enables the inclusion in the user project only of the used features.
( sorry to insist on this, but I don't see any feed-back on this topic )

vgheo, 2016-10-01:
On (2) "CDT configuration"
Unfortunately, this cannot apply for CDT at least, as the an eclipse project is stored as a file hierarchy, where the root directory holds the configuration files
- .project
-.cproject
-.settings

On (3) - "Optional features"
The solution with the compiler flags seems viable, but I have not seen it applied in our projects.
If it is a "new" technique, we need to provide evidence that it is safe.
The gcc version that is valid to use for safety-critical applications has passed some validation conditions (tests/analyses). There are mandatory constraints on gcc usage that an application development configuration must fulfill. 
I am just not aware whether the proposed -Wl,-gc-sections gcc  option is valid to use in our case.
If it is then fine, if not it means extra tool validation effort (which could be fine as well).

ttsiodras, 2016-10-01:
1. Regarding CDT

To be honest, that sounds like bad engineering to me... I gather then, that "out of tree" builds is something impossible for Eclipse projects? All your build cruft simply *must* "pollute" your source tree?

l am more inclined to remove the 3 silly MS RTL build files (.sln, etc - it's not like we use them anyway) - as opposed to to adding more "build-system" pollution in the RTL source folder.  But I will concede to adding the CDT build files in a contrib/CDT folder, with a README explaining how to build the RTL from inside Eclipse (i.e. telling the CDT users to copy wherever you need it to be - I guess in the root). I'd merge that, so that your CDT work is not lost - but as I said, under contrib/CDT.
 
2. Regarding RTL "breakdown" and dead-code detection

I am fully aware of validated compilers (I work for the European Space Agency - we call them "qualified" - as in, qualified for flight). I would also bet that your qualified compilers support the options I told you about (they literally form a "baseline" feature):  the `-fdata-sections -ffunction-sections` tell the compiler to create individual sections (just as the common ones - `.text`, `.data`, `.rodata`, etc) - with each one hosting individual, standalone function object code. Later, in the link stage, the `-Wl,-gc-sections` instructs the linker to check each section to see if there's any other one that needs any of this one's symbols - and if not, it's simply dropped. 

There are many projects where I've seen this approach in action, and it basically "cleans" your binary from any dead code. Any object code left is there because someone needs it.

I therefore see no reason for this breakdown... Not only does it require investing the necessary effort to split things properly, but it will also break existing asn1scc-using codebases (i.e. Makefiles and build processes that depend on the current set of files of the RTL). You do have a point about restructuring for clarity; but honestly, the RTL is quite simple - the real complexity is in the compiler itself (especially when using ACN).

3. For the command line option

Given what you described (your usage scenario), I understood that you want to be able to invoke ASN1SCC without always generating the RTL (since you have many grammars and don't want to have copies of the same file (acn.c, etc). That's why I proposed a command-line flag to NOT emit the RTL (and just emit the grammar-driven files).

Hope this clarifies things - if you are OK with the above, you can send your pull requests (according to the specs in this discussion) and if they are following what we've said in our messages above, I'll merge them in.

maxime-esa, 2020-06-06:
The runtime has now been split into several files as requested in the ticket and only the needed ones are generated by asn1scc.

closed 2020-06-06 by maxime-esa