esa/asn1scc #71: Fails to build on Mac OS X 10.12.1

mouse07410, 2016-11-08:
Mac OS X 10.12.1, Xcode-8.1. Installed Mono-3.12.1. Cloned the current asn1scc master. Followed the instructions - tried to run `./build.sh`:
```
$ ./build.sh 
XBuild Engine Version 12.0
Mono, Version 3.12.1.0
Copyright (C) 2005-2013 Various Mono authors

Build started 11/7/2016 11:38:15 PM.
__________________________________________________
Project "/Users/uri/src/asn1scc/parseStg2/parseStg2.csproj" (default target(s)):
	Target PrepareForBuild:
		Configuration: Debug Platform: x86
		Created directory "bin/Debug/"
		Created directory "obj/x86/Debug/"
	Target GenerateSatelliteAssemblies:
	No input files were specified for target GenerateSatelliteAssemblies, skipping.
	Target CoreCompile:
		Tool /opt/local/lib/mono/4.5/mcs.exe execution started with arguments: /noconfig /debug:full /debug+ /optimize- /out:obj/x86/Debug/parseStg2.exe Program.cs Properties/AssemblyInfo.cs obj/x86/Debug/.NETFramework,Version=v4.0.AssemblyAttribute.cs /target:exe /define:"DEBUG;TRACE" /nostdlib /platform:x86 /reference:/opt/local/lib/mono/4.0/System.dll /reference:/opt/local/lib/mono/4.0/System.Xml.Linq.dll /reference:/opt/local/lib/mono/4.0/System.Data.DataSetExtensions.dll /reference:/opt/local/lib/mono/4.0/Microsoft.CSharp.dll /reference:/opt/local/lib/mono/4.0/System.Data.dll /reference:/opt/local/lib/mono/4.0/System.Xml.dll /reference:/opt/local/lib/mono/4.0/System.Core.dll /reference:/opt/local/lib/mono/4.0/mscorlib.dll /warn:4
	Target DeployOutputFiles:
		Copying file from '/Users/uri/src/asn1scc/parseStg2/obj/x86/Debug/parseStg2.exe.mdb' to '/Users/uri/src/asn1scc/parseStg2/bin/Debug/parseStg2.exe.mdb'
		Copying file from '/Users/uri/src/asn1scc/parseStg2/obj/x86/Debug/parseStg2.exe' to '/Users/uri/src/asn1scc/parseStg2/bin/Debug/parseStg2.exe'
Done building project "/Users/uri/src/asn1scc/parseStg2/parseStg2.csproj".

Build succeeded.
	 0 Warning(s)
	 0 Error(s)

Time Elapsed 00:00:00.9982800
XBuild Engine Version 12.0
Mono, Version 3.12.1.0
Copyright (C) 2005-2013 Various Mono authors

Build started 11/7/2016 11:38:17 PM.
__________________________________________________
Project "/Users/uri/src/asn1scc/Antlr/Antlr.csproj" (default target(s)):
	Target PrepareForBuild:
		Configuration: Debug Platform: AnyCPU
		Created directory "bin/Debug/"
		Created directory "obj/Debug/"
	Target PreBuildEvent:
		Executing: java -jar /Users/uri/src/asn1scc/Antlr/antlr313/antlr-3.2.jar -o /Users/uri/src/asn1scc/Antlr/ /Users/uri/src/asn1scc/Antlr/asn1.g
		java -jar /Users/uri/src/asn1scc/Antlr/antlr313/antlr-3.2.jar -o /Users/uri/src/asn1scc/Antlr/ /Users/uri/src/asn1scc/Antlr/acn.g
		warning(200): /Users/uri/src/asn1scc/Antlr/asn1.g:487:2: Decision can match input such as "{UID, NULL..BIT, BOOLEAN..ENUMERATED, INTEGER..SET, OCTET..UTCTime, OBJECT, RELATIVE_OID}" using multiple alternatives: 1, 2
		As a result, alternative(s) 2 were disabled for that input
		warning(200): /Users/uri/src/asn1scc/Antlr/asn1.g:515:7: Decision can match input such as "L_BRACKET LID {R_BRACKET, COMMA, ':'}" using multiple alternatives: 16, 17
		As a result, alternative(s) 17 were disabled for that input
		warning(200): /Users/uri/src/asn1scc/Antlr/asn1.g:515:7: Decision can match input such as "L_BRACKET LID LID {R_BRACKET, ':'}" using multiple alternatives: 15, 17
		As a result, alternative(s) 17 were disabled for that input
		warning(200): /Users/uri/src/asn1scc/Antlr/acn.g:170:2: Decision can match input such as "SIZE NULL_TERMINATED" using multiple alternatives: 1, 4
		As a result, alternative(s) 4 were disabled for that input
	Target CopyFilesMarkedCopyLocal:
		Copying file from '/Users/uri/src/asn1scc/Antlr/antlr313/antlr.runtime.dll' to '/Users/uri/src/asn1scc/Antlr/bin/Debug/antlr.runtime.dll'
		Copying file from '/Users/uri/src/asn1scc/Antlr/antlr313/Antlr3.Runtime.dll' to '/Users/uri/src/asn1scc/Antlr/bin/Debug/Antlr3.Runtime.dll'
		Copying file from '/Users/uri/src/asn1scc/Antlr/antlr313/Antlr3.Utility.dll' to '/Users/uri/src/asn1scc/Antlr/bin/Debug/Antlr3.Utility.dll'
		Copying file from '/Users/uri/src/asn1scc/Antlr/antlr313/StringTemplate.dll' to '/Users/uri/src/asn1scc/Antlr/bin/Debug/StringTemplate.dll'
	Target GenerateSatelliteAssemblies:
	No input files were specified for target GenerateSatelliteAssemblies, skipping.
	Target CoreCompile:
		Tool /opt/local/lib/mono/4.5/mcs.exe execution started with arguments: /noconfig /debug:full /debug+ /optimize- /out:obj/Debug/Antlr.dll acnLexer.cs acnParser.cs asn1Lexer.cs asn1Parser.cs Comments.cs Properties/AssemblyInfo.cs obj/Debug/.NETFramework,Version=v4.0.AssemblyAttribute.cs /target:library /define:"DEBUG;TRACE" /nostdlib /reference:antlr313/antlr.runtime.dll /reference:antlr313/Antlr3.Runtime.dll /reference:antlr313/Antlr3.Utility.dll /reference:antlr313/StringTemplate.dll /reference:/opt/local/lib/mono/4.0/System.dll /reference:/opt/local/lib/mono/4.0/System.Xml.Linq.dll /reference:/opt/local/lib/mono/4.0/System.Data.DataSetExtensions.dll /reference:/opt/local/lib/mono/4.0/Microsoft.CSharp.dll /reference:/opt/local/lib/mono/4.0/System.Data.dll /reference:/opt/local/lib/mono/4.0/System.Xml.dll /reference:/opt/local/lib/mono/4.0/System.Core.dll /reference:/opt/local/lib/mono/4.0/mscorlib.dll /warn:4
	Target DeployOutputFiles:
		Copying file from '/Users/uri/src/asn1scc/Antlr/obj/Debug/Antlr.dll.mdb' to '/Users/uri/src/asn1scc/Antlr/bin/Debug/Antlr.dll.mdb'
		Copying file from '/Users/uri/src/asn1scc/Antlr/obj/Debug/Antlr.dll' to '/Users/uri/src/asn1scc/Antlr/bin/Debug/Antlr.dll'
Done building project "/Users/uri/src/asn1scc/Antlr/Antlr.csproj".

Build succeeded.
	 0 Warning(s)
	 0 Error(s)

Time Elapsed 00:00:05.8285770
XBuild Engine Version 12.0
[... 39 lines cut ...]
	Task "MSBuild" execution -- FAILED
	Done building target "Build" in project "/Users/uri/src/asn1scc/Asn1.sln".-- FAILED
Done building project "/Users/uri/src/asn1scc/Asn1.sln".-- FAILED

Build FAILED.

Warnings:

/Users/uri/src/asn1scc/Asn1.sln:  warning : Ignoring vcproj 'asn1crt'.
/Users/uri/src/asn1scc/Asn1.sln:  warning : Failed to find project 6094fc85-8ce9-4716-9255-41c428f3b9ea
/Users/uri/src/asn1scc/Asn1.sln:  warning : Don't know how to handle GlobalSection TestCaseManagementSettings, Ignoring.

Errors:

/Users/uri/src/asn1scc/Asn1.sln (default targets) ->
(Build target) ->
/Users/uri/src/asn1scc/AntlrUtils/AntlrUtils.fsproj (default targets) ->

	/Users/uri/src/asn1scc/AntlrUtils/AntlrUtils.fsproj: error : Target named 'Build' not found in the project.

	 3 Warning(s)
	 1 Error(s)

Time Elapsed 00:00:05.4389260
```

ttsiodras, 2016-11-08:
I no longer have access to an OSX machine, unfortunately.

Can you try invoking xbuild like this?

    xbuild /p:TargetFrameworkVersion="v4.6"

mouse07410, 2016-11-09:
```
$ xbuild /p:TargetFrameworkVersion="v4.6"
XBuild Engine Version 12.0
Mono, Version 3.12.1.0
Copyright (C) 2005-2013 Various Mono authors

Build started 11/8/2016 9:19:58 PM.
__________________________________________________
/Users/uri/src/asn1scc/Asn1.sln:  warning : Ignoring vcproj 'asn1crt'.
/Users/uri/src/asn1scc/Asn1.sln:  warning : Failed to find project 6094fc85-8ce9-4716-9255-41c428f3b9ea
/Users/uri/src/asn1scc/Asn1.sln:  warning : Don't know how to handle GlobalSection TestCaseManagementSettings, Ignoring.
Project "/Users/uri/src/asn1scc/Asn1.sln" (default target(s)):
	Target ValidateSolutionConfiguration:
		Building solution configuration "Debug|Mixed Platforms".
	Target Build:
		Project "/Users/uri/src/asn1scc/Antlr/Antlr.csproj" (default target(s)):
			Target PrepareForBuild:
				Configuration: Debug Platform: AnyCPU
			Target GetFrameworkPaths:
/opt/local/lib/mono/4.5/Microsoft.Common.targets:  warning : TargetFrameworkVersion 'v4.6' not supported by this toolset (ToolsVersion: 4.0).
			Target GetReferenceAssemblyPaths:
/opt/local/lib/mono/4.5/Microsoft.Common.targets:  warning : Unable to find framework corresponding to the target framework moniker '.NETFramework,Version=v4.6'. Framework assembly references will be resolved from the GAC, which might not be the intended behavior.
			Target PreBuildEvent:
				Executing: java -jar /Users/uri/src/asn1scc/Antlr/antlr313/antlr-3.2.jar -o /Users/uri/src/asn1scc/Antlr/ /Users/uri/src/asn1scc/Antlr/asn1.g
				java -jar /Users/uri/src/asn1scc/Antlr/antlr313/antlr-3.2.jar -o /Users/uri/src/asn1scc/Antlr/ /Users/uri/src/asn1scc/Antlr/acn.g
				warning(200): /Users/uri/src/asn1scc/Antlr/asn1.g:487:2: Decision can match input such as "{UID, NULL..BIT, BOOLEAN..ENUMERATED, INTEGER..SET, OCTET..UTCTime, OBJECT, RELATIVE_OID}" using multiple alternatives: 1, 2
				As a result, alternative(s) 2 were disabled for that input
				warning(200): /Users/uri/src/asn1scc/Antlr/asn1.g:515:7: Decision can match input such as "L_BRACKET LID {R_BRACKET, COMMA, ':'}" using multiple alternatives: 16, 17
				As a result, alternative(s) 17 were disabled for that input
				warning(200): /Users/uri/src/asn1scc/Antlr/asn1.g:515:7: Decision can match input such as "L_BRACKET LID LID {R_BRACKET, ':'}" using multiple alternatives: 15, 17
				As a result, alternative(s) 17 were disabled for that input
				warning(200): /Users/uri/src/asn1scc/Antlr/acn.g:170:2: Decision can match input such as "SIZE NULL_TERMINATED" using multiple alternatives: 1, 4
				As a result, alternative(s) 4 were disabled for that input
			Target GenerateSatelliteAssemblies:
			No input files were specified for target GenerateSatelliteAssemblies, skipping.
			Target CoreCompile:
				Tool /opt/local/lib/mono/4.5/mcs.exe execution started with arguments: /noconfig /debug:full /debug+ /optimize- /out:obj/Debug/Antlr.dll acnLexer.cs acnParser.cs asn1Lexer.cs asn1Parser.cs Comments.cs Properties/AssemblyInfo.cs obj/Debug/.NETFramework,Version=v4.6.AssemblyAttribute.cs /target:library /define:"DEBUG;TRACE" /nostdlib /reference:antlr313/antlr.runtime.dll /reference:antlr313/Antlr3.Runtime.dll /reference:antlr313/Antlr3.Utility.dll /reference:antlr313/StringTemplate.dll /reference:/opt/local/lib/mono/gac/System/4.0.0.0__b77a5c561934e089/System.dll /reference:/opt/local/lib/mono/gac/System.Xml.Linq/4.0.0.0__b77a5c561934e089/System.Xml.Linq.dll /reference:/opt/local/lib/mono/gac/System.Data.DataSetExtensions/4.0.0.0__b77a5c561934e089/System.Data.DataSetExtensions.dll /reference:/opt/local/lib/mono/gac/Microsoft.CSharp/4.0.0.0__b03f5f7f11d50a3a/Microsoft.CSharp.dll /reference:/opt/local/lib/mono/gac/System.Data/4.0.0.0__b77a5c561934e089/System.Data.dll /reference:/opt/local/lib/mono/gac/System.Xml/4.0.0.0__b77a5c561934e089/System.Xml.dll /reference:/opt/local/lib/mono/gac/System.Core/4.0.0.0__b77a5c561934e089/System.Core.dll /warn:4
CSC: error CS0518: The predefined type `System.Object' is not defined or imported
CSC: error CS0518: The predefined type `System.ValueType' is not defined or imported
CSC: error CS0518: The predefined type `System.Attribute' is not defined or imported
CSC: error CS0518: The predefined type `System.Int32' is not defined or imported
CSC: error CS0518: The predefined type `System.UInt32' is not defined or imported
CSC: error CS0518: The predefined type `System.Int64' is not defined or imported
CSC: error CS0518: The predefined type `System.UInt64' is not defined or imported
CSC: error CS0518: The predefined type `System.Single' is not defined or imported
CSC: error CS0518: The predefined type `System.Double' is not defined or imported
CSC: error CS0518: The predefined type `System.Char' is not defined or imported
CSC: error CS0518: The predefined type `System.Int16' is not defined or imported
CSC: error CS0518: The predefined type `System.Decimal' is not defined or imported
CSC: error CS0518: The predefined type `System.Boolean' is not defined or imported
CSC: error CS0518: The predefined type `System.SByte' is not defined or imported
CSC: error CS0518: The predefined type `System.Byte' is not defined or imported
CSC: error CS0518: The predefined type `System.UInt16' is not defined or imported
CSC: error CS0518: The predefined type `System.String' is not defined or imported
CSC: error CS0518: The predefined type `System.Enum' is not defined or imported
CSC: error CS0518: The predefined type `System.Delegate' is not defined or imported
[... cut ...]
	CSC: error CS0518: The predefined type `System.RuntimeTypeHandle' is not defined or imported
	CSC: error CS0518: The predefined type `System.Exception' is not defined or imported

/Users/uri/src/asn1scc/Asn1.sln (default targets) ->
(Build target) ->
/Users/uri/src/asn1scc/AntlrUtils/AntlrUtils.fsproj (default targets) ->

	/Users/uri/src/asn1scc/AntlrUtils/AntlrUtils.fsproj: error : Target named 'Build' not found in the project.

	 5 Warning(s)
	 32 Error(s)

Time Elapsed 00:00:24.8065560
```

Are you suggesting that I should try with a different Mono?

maxime-esa, 2016-11-09:
Your log shows "4.5" - did you try with this one?
i..e

```
$ xbuild /p:TargetFrameworkVersion="v4.5"
```

ttsiodras, 2016-11-09:
I don't think it's related to the framework version - the original error reported here drills down to...

    "Project "/Users/uri/src/asn1scc/AntlrUtils/AntlrUtils.fsproj" (default target(s)):
    /Users/uri/src/asn1scc/AntlrUtils/AntlrUtils.fsproj: error : Target named 'Build' not found in the project."

I also built in my Arch linux, using the latest mono (4.6) - it worked there as well.
It seems that mono 3.12.1.0 (which is the one @mouse07410 is using, had a regression in `xbuild`.

@mouse07410 - can you install an updated (or an older! :-) ) version of Mono?

mouse07410, 2016-11-09:
Installing the current Mono from http://www.mono-project.com/docs/about-mono/supported-platforms/osx/ resolved this problem. Therefore the bug indeed lies with the Macports mono version. 

```
mono Asn1f2/bin/Debug/Asn1f2.exe
No input files

Semantix ASN.1 Compiler
Current Version is: 3.3.07 
Usage:

asn1  <OPTIONS> file1, file2, ..., fileN 

Where OPTIONS are:

. . . . .
```

Thanks!

closed 2016-11-09 by mouse07410