esa/asn1scc #77: Duplicate definition report when using parameterized type

shd01x, 2017-04-10:
I have two modules (in test.asn and second.asn)

test.asn contains:

```
----------------------------------------------------------------------------------------
-- Test module
----------------------------------------------------------------------------------------
Test-Dataview DEFINITIONS ::=
BEGIN

IMPORTS 
  T-SecondModule-Type1 FROM SecondModule-Dataview;

T-Test{TypeParameter} ::= SEQUENCE {
  type1Thing            T-SecondModule-Type1,
  value                 TypeParameter
}  

T-MyParameterisedType  ::= T-Test{INTEGER}

END
```
second.asn contains:

```
----------------------------------------------------------------------------------------
-- Second module
----------------------------------------------------------------------------------------

SecondModule-Dataview DEFINITIONS ::=
BEGIN

T-SecondModule-Type1                ::= INTEGER

END

```
When I run the following command:

`asn1.exe test.asn second.asn `

I get the following error report:

`Duplicate definition: T-SecondModule-Type1`

I don't understand why I get this error.  Am I doing something wrong?

shd01x, 2017-04-10:
Should have said, version 3.3.09 running in the TASTE VM.

maxime-esa, 2017-04-11:
@shd01x : I have compiled the latest version (3.3.18) and updated it in the VM. If you run Update-TASTE.sh this new version will be installed and you can try again

Thanks George for the fix

shd01x, 2017-04-11:
Ah. I spoke too soon.  

My original test case passes, but if I add a third module the problem reappears:

```
----------------------------------------------------------------------------------------
-- Test module
----------------------------------------------------------------------------------------
Test-Dataview DEFINITIONS ::=
BEGIN

IMPORTS 
  T-SecondModule-Type1 FROM SecondModule-Dataview;

T-Test ::= SEQUENCE {
  type1Thing            T-SecondModule-Type1{INTEGER}
}  

END

----------------------------------------------------------------------------------------
-- Second module
----------------------------------------------------------------------------------------

SecondModule-Dataview DEFINITIONS ::=
BEGIN

IMPORTS
  T-ThirdModule-Type2  FROM ThirdModule-Dataview;

T-SecondModule-Type3 ::= INTEGER
T-SecondModule-Type4 ::= INTEGER

T-SecondModule-Type1{TypeParameter} ::= SEQUENCE {
  val1                  T-SecondModule-Type3  ,
  val2                  T-ThirdModule-Type2  ,
  val3                  T-SecondModule-Type4 ,
  value                 TypeParameter
}  

END

----------------------------------------------------------------------------------------
-- Third module
----------------------------------------------------------------------------------------

ThirdModule-Dataview DEFINITIONS ::=
BEGIN

T-ThirdModule-Type2 ::= INTEGER(0 .. 33)

END

```

Output of asn1scc:

```
assert@tastevm:~/asn1scc_bug$ asn1.exe test.asn second.asn third.asn
Duplicate definition: T-SecondModule-Type3

```

gmamais, 2017-04-12:
Please try again the latest version 3.19

maxime-esa, 2017-04-12:
Thanks, compiled in the TASTE VM, run again Update-TASTE.sh to get this new version

shd01x, 2017-04-12:
Awesome!  Both the above test cases, and my mad ASN.1 model, now compile perfectly.

Thanks again guys for your very quick work on this.  Will buy the beers next time I'm at ESTEC :)

closed 2017-04-12 by maxime-esa