esa/asn1scc #78: "Conflicting types" GCC error in autogenerated code for value assignment

shd01x, 2017-04-11:
Running 3.3.9 in TASTE VM, the ASN.1 compiler emits some illegal code.

The following ASN.1 module:

```
----------------------------------------------------------------------------------------
-- Test module
----------------------------------------------------------------------------------------
Test-Dataview DEFINITIONS ::=
BEGIN

T-Test ::= SEQUENCE {
  type1Thing           INTEGER,
  value                INTEGER
}  

c-A-Thing SEQUENCE SIZE(1 .. 10) OF T-Test ::= {
  {type1Thing 0, value 1}
}

END

```
gives the following error when compiled:

```
assert@tastevm:~/asn1scc_bug2$ . asn1_compile 
assert@tastevm:~/asn1scc_bug2$ gcc -c test.c
test.c:40:3: error: conflicting types for 'c_A_Thing'
 } c_A_Thing = {    .nCount = 1,    .arr = 
   ^
In file included from test.c:7:0:
test.h:32:3: note: previous declaration of 'c_A_Thing' was here
 } c_A_Thing; 
   ^
```

The reason is that C treats both anonymous structure definitions (i.e. the definition in the C file and the extern declaration in the H file) as new types:

> §6.7.2.3/p5: "Each declaration of a structure, union, or enumerated type which does not include a tag declares a distinct type."

maxime-esa, 2017-04-11:
Confirmed, this bug is already identified in the TASTE bugtrack, here:
http://taste.tuxfamily.org/mantis/view.php?id=516

The constant declaration of a variable of an anonymous type should be forbidden (ASN.1 compiler to raise an error)

shd01x, 2017-04-11:
Thanks Maxime for your quick reply :)  

By "anonymous type" I think you refer to the variable length of my SEQUENCE OF.  But I think the problem is greater than this - it still happens when I define a sequence of a fixed length, e.g.

```
c-A-Thing SEQUENCE SIZE(1) OF T-Test ::= {
  {type1Thing 0, value 1}
}
```
Nothing anonymous about this, but still fails to compile :(

A good (and quick?) alternative solution could be to give the struct a tag name in the H file and declare a variable of that tagged struct type.

BTW should I be posting my observations in the TASTE bugtrack instead of here?

Steve D

shd01x, 2017-04-11:
In case it helps, I was able to proceed after the following emergency surgery on ch.stg

```
Declare_SequenceOf(bIsFixedSize, sTypeDecl, nLength, sArrayPostfix) ::= <<

#ifndef <sTypeDecl>_s_DEFINED
#define <sTypeDecl>_s_DEFINED
struct <sTypeDecl>_s {<if(!bIsFixedSize)>
    int nCount; 

<endif>    
    <sTypeDecl> arr[<nLength>]<sArrayPostfix>;
}
#else
struct <sTypeDecl>_s
#endif  

>>
```

NB the new blank lines are mandatory.

gmamais, 2017-04-11:
Hi,
By Anonymous types, we mean when declaring variable and the type at the same line.
Instead of writing:
c-A-Thing SEQUENCE SIZE(1) OF T-Test ::= {
  {type1Thing 0, value 1}
}
Write this
MySeq ::= SEQUENCE SIZE(1) OF T-Test
c-A-Thing MySeq ::= {
  {type1Thing 0, value 1}
}

shd01x, 2017-04-11:
Oh, I see.  Thanks for clarifying :)

closed 2018-05-19 by maxime-esa