esa/asn1scc #79: Code generation for WITH COMPONENTS

shd01x, 2017-04-18:
If a sequence with optional components is fully constrained by a WITH COMPONENTS clause, the autogenerated type nevertheless contains an "exist" field, and absent components are still encoded.

For example, the following ASN.1

```
Test DEFINITIONS ::=

BEGIN

T-Test ::= SEQUENCE {
  field1 INTEGER,
  field2 INTEGER,
  field3 INTEGER,
  field4 INTEGER,
  field5 INTEGER
}

T-Test2 ::= SEQUENCE {
  field1 INTEGER OPTIONAL,
  field2 INTEGER OPTIONAL,
  field3 INTEGER OPTIONAL,
  field4 INTEGER OPTIONAL,
  field5 INTEGER OPTIONAL
}

T-Test3 ::= T-Test2 ( WITH COMPONENTS {
  field1 PRESENT,
  field2 PRESENT,
  field3 ABSENT,
  field4 ABSENT,
  field5 PRESENT
  } )
  
END
  
```
generates these types:

```

typedef struct {
    asn1SccSint field1;
    asn1SccSint field2;
    asn1SccSint field3;
    asn1SccSint field4;
    asn1SccSint field5;
} T_Test;

#define T_Test_REQUIRED_BYTES_FOR_ENCODING       45 

typedef struct {
    asn1SccSint field1;
    asn1SccSint field2;
    asn1SccSint field3;
    asn1SccSint field4;
    asn1SccSint field5;
    struct {
        unsigned int field1:1;
        unsigned int field2:1;
        unsigned int field3:1;
        unsigned int field4:1;
        unsigned int field5:1;
    } exist;
} T_Test2;

#define T_Test2_REQUIRED_BYTES_FOR_ENCODING       46 

typedef struct {
    asn1SccSint field1;
    asn1SccSint field2;
    asn1SccSint field3;
    asn1SccSint field4;
    asn1SccSint field5;
    struct {
        unsigned int field1:1;
        unsigned int field2:1;
        unsigned int field3:1;
        unsigned int field4:1;
        unsigned int field5:1;
    } exist;
} T_Test3_1;

#define T_Test3_1_REQUIRED_BYTES_FOR_ENCODING       46 

```
So space for the fields is still reserved in the encoding even when they are known to be absent, and space for the "exist" field is also unnecessary IMHO.

I would strongly prefer the ABSENT fields not to appear in the C struct.  Please could this be considered?

maxime-esa, 2017-04-18:
This was recently discussed here: http://taste.tuxfamily.org/mantis/view.php?id=543
(Let me know if you do not have an account or face issues creating one/accessing it)

Basically two conclusions were reached:

1. Removing the .exist field could break existing code so we decided to keep it, but the encoding function ignores it. Note that with `uPER` the bit in the encoding packet has to be there in any case, this is a rule of the `uPER` encodings (this subtype constraint is not visible to `uPER`). So even if you have a `WITH COMPONENTS` (`PRESENT` or `ABSENT`), it remains an optional field from the point of view of the encoder.

2. `ACN` was updated to solve this last "issue" by  not putting this extra bit in the packets if there is a subtype constraint

shd01x, 2017-04-18:
Thanks, Maxime.  I searched mantis for this but obviously my search wasn't good enough :(

I will explore what can be done with the ACN and close this issue.

closed 2017-04-18 by shd01x