esa/asn1scc #107: JSON serialization

vgheo, 2017-07-20:
It would be valuable to support JSON serialization.

maxime-esa, 2017-07-21:
There are no JSON standard encoding rules for ASN.1 so this is not possible.

You may look at the ASN.1 [Generic String Encoding Rules](https://tools.ietf.org/html/rfc3641) instead, which are fairly easy to implement, and compliant to the ASN.1 semantics.

But we are not going to add native GSER encoding in the core of ASN1SCC in the near future as we can't really afford to support too many encoding runtimes, especially when they do not target safety-critical systems that need compact binary encodings. 

The options if you want to implement this kind of encodings are the following:

1) use the ASN1SCC Python API from [TASTE Data Modelling tools](https://github.com/ttsiodras/DataModellingTools) to get access to the ASN.1 types. The Python API already provides a GSER() function that encodes in that format... We also already have a basic parser in Python for that format in [TASTE](http://taste.tools). 

2) Encode in XER and convert to JSON. I am pretty sure there exists XML to JSON converters...

3) Also in the DMT tools (see point 1 above) there is an ASN.1 to SQLAlchemy bridge, so you can store ASN.1 values in a SQL Database which structure is compliant to the ASN.1 AST. From this point you can easily interface with JSON.

vgheo, 2017-07-22:
@maxime-esa  - thank you for the valuable information.

In our current context, any ASN1-JSON encoding would be acceptable - even though ideally it should be an international standard.

There is a [proprietary specification by OSSNokalva](http://www.oss.com/asn1/resources/books-whitepapers-pubs/JSON%20Encoding%20Rules.pdf
).
It is unclear what the legal terms of use are for this specification.

Related : [diagram of asn1scc architecture](http://taste.tuxfamily.org/wiki/index.php?title=Technical_topic:_Customizing_the_ASN.1_compiler_(ASN1SCC)_with_your_own_templates)

vgheo, 2017-07-22:
Regarding the maintenance of the JSON encoding runtime - this would amount to the selection of one of the many [JSON encoding/decoding implementations](http://json.org/).

IMO - this means that almost no maintenance is involved : the only maintenance required it to keep the code generator compatible with the API/datamodel of the selected JSON library (which is expected to be quite stable).

maxime-esa, 2017-07-22:
Understood .. Thanks for letting us know if you develop this runtime, you may commit it under the contrib folder and I'll check with the team if it can be merged with the core tool.

closed 2017-07-22 by maxime-esa