esa/asn1scc #108: Generate UML model/diagrams from asn1 files

vgheo, 2017-07-20:
For complex models, a diagram generated from the asn1 files would be a valuable documentation.

vgheo, 2017-07-20:
The xml ast is an obvious choice as a source format.
Is this stable ?
How likely is it to change in the future?

vgheo, 2017-07-20:
One possible target is eclipse UML for model and [Papyrus](https://eclipse.org/papyrus)  for diagrams.

- XSD model converted to ecore
-  model-to-model transform (asn1scc-ast to UML) using [Eclipse QVTo](https://wiki.eclipse.org/QVTo)

User would need a standard Papyrus/eclipse installation plus the transform plugin.

Q: Is the XSD for the AST available ?

vgheo, 2017-07-20:
Another interesting target  is [PlantUML class diagram](http://plantuml.com/class-diagram).

This is a much lighter approach compared to the other one - as no explicit meta-models are involved.

Could be done quite easily as python script. 
xslt could be an option, but it is less flexible and has a smaller developer base.

maxime-esa, 2017-07-20:
The backends are provided in [StringTemplate](http://www.stringtemplate.org) form, including the XML backend that you can trigger with -ast. 

There is also a command line option (`-customStg`) that allows you to use your own backend, based on the XML one (i.e. you can output the AST in any form you like). The Python backend for example is based on this (python.stg).

I think this is what you are looking for. Copy `xml.stg` to `eclipse.stg` and write your own AST generator. This is an easy process. You don't need to recompile the tool.. just invoke it with:
`asn1.exe -customStg eclipse.stg:my_uml_ast.xml datamodel.asn` and you are done!

In terms of evolution, we are working on a large refactoring of some parts of the tool, which will include extension of the XML AST backend (to include ACN content). So there will be impact on existing backends, which will have to be adapted.

vgheo, 2017-07-20:
Thank you for the reply.

I know about parsers, ASTs, model-to-text transforms in general.
Technologies I have used: yacc/bison, xtext ( based on antlr), EMF (for AST model), MTL / acceleo ( model-to-text).

I am completely new to StringTemplate - I will read on.

What I fail to understand right now is the interface between the asn1scc core and the backends.

My expectation is that this interface consists of a pre-determined data model of the inputs ( collection of ASTs).
Also, from my general understanding of the term template (as in XSLT, MTL, JSP), a template language has templates that are user-defined functions from an input (normally a list of instances of a data model) to arbitrary text, where a template may call other template.

Looking at xml.stg, I have the following questions
- Where are the types of parameters (input data model) defined ?
   - I'd like to inspect the input data model (AST) itself rather than guessing it from this particular use 
- Where are these templates called from ? Is there a single entry point? (doesn't look so)
- Is this group of templates stand-alone, or it has some other dependencies ?

Best regards
Vlad

maxime-esa, 2017-07-21:
After parsing the input, the compiler has its own internal representation of the AST (in F#). Unless you study the source code of the compiler, you usually won't look at this one (it can change as the compiler evolves).

The AST is exposed to the user via the template files, which can be defined at will to represent the AST in a user-defined form (which I think is what you want to do). Since the compiler processes the AST, the templates can receive different level of information:

- templates of the family of the `-ast` flag mostly contain the dump of the data type definitions, but is also influenced by some command line flags (_-typePrefix_, _-renamePolicy_)
- templates of the ICD family (`-customIcd`) receive information about the binary encodings
- templates for code generation in C and Ada are also different

Is this information sufficient for you to build your template files?

maxime-esa, 2018-05-19:
Closing if there are no more questions. Feel free to reopen if needed.

maxime-esa, 2020-04-20:
Dear Vlad,

It's late but you might be glad to know that now the XSD of the full XML AST (the one you get with -x flag) is now available and generated together with the XML output.

It's possible easily to represent it with a UML class diagram (this was done in an Eclipse plugin by some users)

closed 2020-06-06 by maxime-esa