esa/asn1scc #109: Crash when partially overriding INTEGER encoding in ACN

hcorg, 2017-10-17:
Although code shown below might not be a good model, it's a bad thing asn1scc crashes, instead of providing some sort of information to the user.

Given ASN.1:
```
Crashing DEFINITIONS AUTOMATIC TAGS ::= BEGIN

SomeInt ::= INTEGER (0 .. 65535)

SomeSequence ::= SEQUENCE
{
  field SomeInt
}

END
```

And ACN:
```
Crashing DEFINITIONS ::= BEGIN

SomeInt [size 16, encoding pos-int]

SomeSequence []
{
  field [size 16] -- here's a line that causes problems, adding encoding here solves problem
}

END
```

when trying to generate code using `asn1.exe -c -ACN Crash.asn1 Crash.acn` causes asn1scc to exit with message `Object reference not set to an instance of an object` which suggests uncaught `System.NullReferenceException`.

hcorg, 2017-10-17:
[Crash_sources.zip](https://github.com/ttsiodras/asn1scc/files/1390745/Crash_sources.zip)

gmamais, 2017-10-17:
This is strange because ASN1SCC ver3 does not allow user to put acn properties in referenced types (i.e. the 'field' component within sequence). 
In any case, the compiler must not crash. 
I will look into it on the weekend.

gmamais, 2017-10-22:
The bug was fixed by simply re-enabling the check which does not allow acn properties in referenced types.
Now the compiler complaints that you cannot put the size property in this location.

hcorg, 2017-11-14:
tested on master:
`Crash.acn:7:9: error: Acn property 'size' cannot be applied here`

closed 2017-11-14 by hcorg