esa/asn1scc #110: Incorrect C code generated from ASN.1/ACN model

hcorg, 2017-10-17:
Given model: [bug.zip](https://github.com/ttsiodras/asn1scc/files/1390798/bug.zip)
after generating code using `asn1.exe -c -ACN Bug.asn1 Bug.acn` and then trying to compile it (`Bug.c`)
compilation fails - `*` used to dereference not a pointer.

Model uses two tiers of types to access field introduced in ACN (`parameterizedType <typeTier0.typeTier1.someType>`)

gmamais, 2017-10-22:
This bug does not appear in the upcoming version of the compiler (V4). Version 4 handles completely different such cases (if you have time, just look at the generated code). Since version 4 will very soon replace the existing version, it will no longer be an issue. Please let me know if this bug is a blocking issue in you project or you can wait for few weeks for the new release.

hcorg, 2017-10-22:
I do not think it's blocking, @michalkocon what do you think?

michalkocon, 2017-10-23:
I already redesigned `SEQUENCE`s not to use so much tiers in parameterization so it's not a problem, for now at least.

closed 2017-11-14 by hcorg