esa/asn1scc #114: SEQUENCE/SEQUENCE OF is indistinguishable from SET/SET OF in AST XML file

psockoN7, 2017-10-27:
Given ASN.1:
```
Module DEFINITIONS AUTOMATIC TAGS ::= BEGIN
    MySet ::= SET { }
    MySequence ::= SEQUENCE { }
END
```
the AST obtained with after invoking: ```Asn1f2.exe -ast ast.xml sequences.asn1``` is as follows:
```
<?xml version="1.0" encoding="utf-8"?>
<ASN1AST>
    <Asn1File FileName="sequences.asn1">
        <Asn1Module ID="Module">
        <ExportedTypes>
            <ExportedType Name="MySet" />
            <ExportedType Name="MySequence" /> 
        </ExportedTypes>
        <ExportedVariables>
         
        </ExportedVariables>
        <ImportedModules>
        </ImportedModules>
        <TypeAssignments>
            <TypeAssignment Name="MySet" Line="2" CharPositionInLine="4">
                <Type Line="2" CharPositionInLine="14">
                    <SequenceType>
                    </SequenceType>
                </Type>
            </TypeAssignment>
            <TypeAssignment Name="MySequence" Line="3" CharPositionInLine="4">
                <Type Line="3" CharPositionInLine="19">
                    <SequenceType>
                    </SequenceType>
                </Type>
            </TypeAssignment>
        </TypeAssignments>
        <VariablesAssignments>
        </VariablesAssignments>
        </Asn1Module>
    </Asn1File>
</ASN1AST>
```
The information about whether type definition is SEQUENCE or SET is missing in produced XML file. It would be a welcome change, if SEQUENCE/SEQUENCE OF could be distinguished from SET/SET OF, based on AST XML file.

ttsiodras, 2017-10-28:
This was a deliberate choice in ASN1SCC; SETs and SEQUENCEs are handled in exactly the same way in all the TASTE code generators.

closed 2017-10-28 by ttsiodras

hcorg, 2017-10-30:
ACK: all SETs will be described as SEQUENCEs in all asn1scc.IDE GUI widgets.