esa/asn1scc #115: Incorrect importing of value assignment, when there are no imported types.

psockoN7, 2017-11-13:
Given file: 
```
Exporter DEFINITIONS AUTOMATIC TAGS ::= BEGIN

EXPORTS ALL;

    val INTEGER ::= 157
END
```

and file:

```
Importer DEFINITIONS AUTOMATIC TAGS ::= BEGIN

IMPORTS val FROM Exporter;

    MyInt ::= INTEGER (-256 .. val)
END
```

produced AST is following (please note that _ImportedModules_ section is empty):
```
<?xml version="1.0" encoding="utf-8"?>
<ASN1AST>
    <Asn1File FileName="Importer.asn">
        <Asn1Module ID="Importer">
        <ExportedTypes>
            <ExportedType Name="MyInt" /> 
        </ExportedTypes>
        <ExportedVariables>
         
        </ExportedVariables>
        <ImportedModules>
        </ImportedModules>
        <TypeAssignments>
            <TypeAssignment Name="MyInt" Line="5" CharPositionInLine="4">
                <Type Line="5" CharPositionInLine="14">
                    <IntegerType Min="-256" Max="157"/>
                </Type>
            </TypeAssignment>
        </TypeAssignments>
        <VariablesAssignments>
        </VariablesAssignments>
        </Asn1Module>
    </Asn1File>
    <Asn1File FileName="Exporter.asn">
        <Asn1Module ID="Exporter">
        <ExportedTypes>
         
        </ExportedTypes>
        <ExportedVariables>
            <ExportedVariable Name="val" /> 
        </ExportedVariables>
        <ImportedModules>
        </ImportedModules>
        <TypeAssignments>
        </TypeAssignments>
        <VariablesAssignments>
            <VariableAssignment Name="val" Line="5" CharPositionInLine="4">
                <Type Line="5" CharPositionInLine="8">
                    <IntegerType Min="MIN" Max="MAX"/>
                </Type>
                <Value>157</Value>
            </VariableAssignment>
        </VariablesAssignments>
        </Asn1Module>
    </Asn1File>
</ASN1AST>
```

Produced C code is also invalid for that case.

**However**, given file:
```
Exporter DEFINITIONS AUTOMATIC TAGS ::= BEGIN

EXPORTS ALL;

    val INTEGER ::= 157
    IntType ::= INTEGER
END
```

and file:
```
Importer DEFINITIONS AUTOMATIC TAGS ::= BEGIN

IMPORTS val, IntType FROM Exporter;

    MyInt ::= INTEGER (-256 .. val)
END
```

produced AST is following (please note _ImportedModules_ section is not empty):

```
<?xml version="1.0" encoding="utf-8"?>
<ASN1AST>
    <Asn1File FileName="Importer.asn">
        <Asn1Module ID="Importer">
        <ExportedTypes>
            <ExportedType Name="MyInt" />
            <ExportedType Name="IntType" /> 
        </ExportedTypes>
        <ExportedVariables>
         
        </ExportedVariables>
        <ImportedModules>
            <ImportedModule ID="Exporter">
                <ImportedTypes>
                    <ImportedType Name="IntType" />
                </ImportedTypes>
                <ImportedVariables>
                    <ImportedVariable Name="val" />
                </ImportedVariables>
            </ImportedModule>
        </ImportedModules>
        <TypeAssignments>
            <TypeAssignment Name="MyInt" Line="5" CharPositionInLine="4">
                <Type Line="5" CharPositionInLine="14">
                    <IntegerType Min="-256" Max="157"/>
                </Type>
            </TypeAssignment>
        </TypeAssignments>
        <VariablesAssignments>
        </VariablesAssignments>
        </Asn1Module>
    </Asn1File>
    <Asn1File FileName="Exporter.asn">
        <Asn1Module ID="Exporter">
        <ExportedTypes>
            <ExportedType Name="IntType" /> 
        </ExportedTypes>
        <ExportedVariables>
            <ExportedVariable Name="val" /> 
        </ExportedVariables>
        <ImportedModules>
        </ImportedModules>
        <TypeAssignments>
            <TypeAssignment Name="IntType" Line="6" CharPositionInLine="4">
                <Type Line="6" CharPositionInLine="16">
                    <IntegerType Min="MIN" Max="MAX"/>
                </Type>
            </TypeAssignment>
        </TypeAssignments>
        <VariablesAssignments>
            <VariableAssignment Name="val" Line="5" CharPositionInLine="4">
                <Type Line="5" CharPositionInLine="8">
                    <IntegerType Min="MIN" Max="MAX"/>
                </Type>
                <Value>157</Value>
            </VariableAssignment>
        </VariablesAssignments>
        </Asn1Module>
    </Asn1File>
</ASN1AST>
```

Produced C code is valid in this case.

hcorg, 2018-01-31:
@gmamais @maxime-esa same problem in V4 - when importing *only* value from other module it does not appear in `ImportedModules`.

C code seems to be ok in V4 though.

gmamais, 2018-06-09:
This issue has been closed by
https://github.com/ttsiodras/asn1scc/pull/137

PR #137 (2018-02-02): [V4] ImportedModules will appear in XML AST even when only values are imported
This closes #115

Also - I've removed some code duplication - in `MapParamAstToNonParamAst.fs` imported modules have already been filtered for parametrized types.
commit edf38c92 (2018-02-02): ImportedModules will appear in XML AST even when only values are imported

This closes #115
closed 2018-06-09 by gmamais