esa/asn1scc #129: [V4][XML.AST] <CONS> and <WITH_CONS> items are not self-explanatory

hcorg, 2018-01-29:
I guess, that <CONS> are really <Constraints> :) but I have no idea what <WITH_CONS /> is - maybe better name could help future users.

gmamais, 2018-01-30:
Ok, I will rename CONS to Constraints
and WITH_CONS to WithComponentConstraint.
WITH COMPONENTS constraints  are handled differently because they have no impact on UPER encoding.
e.g.

MYINT ::= (0..1023) -- 10 bits
MYSEQOF ::= SEQUENCE (SIZE(10)) OF MYINT (WITH COMPONT (0..31)) --MYSEQOF will be encoded as 10*10bits = 100 bits even though the allowed values for the integer is 0..31

hcorg, 2018-01-30:
ooooh, I completely forgot about `WITH COMPONENTS` :D

gmamais, 2018-01-30:
CONS renamed to Constraints
and 
WITH_CONS to WithComponentConstraints

closed 2018-01-30 by gmamais