esa/asn1scc #132: [V4][XML.AST] Suggestion: <a> and <b> in <Range> could be named better

hcorg, 2018-01-29:
<begin><end>?
<min><max>?
<beginIncluded><endIncluded>?

it's not big change, but <a> and <b> suggest <c> IMHO ;)

maxime-esa, 2018-01-30:
A range can be more than min and max. For example
```
MyInt ::= INTEGER (0..255 | (512..1024) | 10)
```
In V3 the XML backend was only giving Min and Max but that was incomplete.

hcorg, 2018-01-30:
ok, but multiple ranges look like that:
```
<CONS>
                  <OR>
                    <OR>
                      <Range>
                        <a>
                          <IntegerValue>0</IntegerValue>
                        </a>
                        <b>
                          <IntegerValue>255</IntegerValue>
                        </b>
                      </Range>
                      <IntegerValue>333</IntegerValue>
                    </OR>
                    <Range>
                      <a>
                        <IntegerValue>512</IntegerValue>
                      </a>
                      <b>
                        <IntegerValue>1024</IntegerValue>
                      </b>
                    </Range>
                  </OR>
                </CONS>
```

hcorg, 2018-05-21:
Fixes in our code will be simple and it will not break any existing software (as it has asn1scc version embedded in itself)

gmamais, 2018-05-26:
a changed to Min
b changed to Max

closed 2018-05-26 by gmamais