esa/asn1scc #145: Linux build fails

psockoN7, 2018-04-18:
When building project on Linux platform, following error is reported
```
Service.Implementation/Asn1Service.fs(87,36): error FS0001: This expression was expected to have type    'a * 'b    but here has type    (Asn1Ast.AstRoot -> AcnCreateFromAntlr.AcnAst -> unit) -> Asn1AcnAst.AstRoot * Asn1AcnAst.AcnInsertedFieldDependencies
```
which terminates build.

HEAD is : `2bbc66af93ca6be25c26bcecece47f8b6f99ee4a`

OS information:
```
Operating System: Debian GNU/Linux 9 (stretch)
Kernel: Linux 4.9.0-6-amd64
Architecture: x86-64
```

hcorg, 2018-04-18:
failing line _probably_ needs to be changed into following two lines:
```
 let debugFunc (r:Asn1Ast.AstRoot) (acn:AcnCreateFromAntlr.AcnAst) = ()
 let frontEntAst, acnDeps = FrontEntMain.constructAst args debugFunc
```

I can't check it quickly, as project still does not compile due to some missing files... (PrintAcn.fs)

gmamais, 2018-04-18:
My bad ... (I forgot to add the file.)
Added.

closed 2018-04-18 by gmamais