esa/asn1scc #152: Bit order in Ada

gregerts, 2018-05-18:
I am testing the latest version of ASN1CC with GNAT Pro on the ARM Cortex R5 (_little-endian_) using ACN format for _big-endian_ unsigned 16-bit integers. However, I am experiencing a problem with the expected bit order of the code generated by the compiler. The bytes: `0x02 0x03` should decode to the integer `0x0203`, but is instead decoded as `0x40C0`. If you look at the bits you can see that each byte has a **reversed bit order**. I assume that this is resulted by the compiler formatting the packed Boolean array with byte alignment, and the bytes in the expected order, but **within each byte the lowest index is the least significant bit**, not the most significant as expected by the code:

    PROCEDURE BitStream_Decode_Non_Negative_Integer (S : in BitArray;
                                                     P : in out DECODE_PARAMS;
                                                     IntValue : out     Asn1UInt;
                                                     nBitsRange : IN Integer;
                                                     result : OUT Boolean)
    --# derives IntValue from S, P, nBitsRange & P from P, nBitsRange & result from P, nBitsRange;
    --# pre 	nBitsRange >= 0 and nBitsRange <= 64 and
    --#         P.K+1>= S'First and P.K + nBitsRange <= S'Last;
    --# post P.K = P~.K + nBitsRange;
    IS
    BEGIN

        IntValue := 0;
        FOR I IN Integer range 1..nBitsRange LOOP
    --# assert 	nBitsRange >= 1 and nBitsRange <= 64 and
    --#         P.K + nBitsRange <= S'Last and
    --#         P.K+1>= S'First and
    --#		P.K=P~.K and  P.K + I - 1< S'Last;

            IF S(P.K+I)=0 THEN
                IntValue := 2*IntValue;
            ELSE
                IntValue := 2*IntValue + 1;
            END IF;

        END LOOP;

        P.K := P.K + nBitsRange;
        result := P.DataLen-P.K>=0;
    END BitStream_Decode_Non_Negative_Integer;

Am I missing something here? This seems to be the standard way of GNAT to output packed Boolean arrays. The C output uses the right bit order. Also, the Ada code above is not very efficient, the C codes use of bytes from the buffer seems better.

gregerts, 2018-05-18:
I also experience the same error on PC where the Ada and C code produce different bytes for the same ACN and the same data. See attached code.
[asn1cc-bit-order.zip](https://github.com/ttsiodras/asn1scc/files/2016677/asn1cc-bit-order.zip)

gmamais, 2018-05-18:
The problem is in the way you dump the bitstream into a file.

If you encode the value 0x0203 then bits should be sent to the other side as follows

0000 0010 0000 0011

This exactly the contents of the bitstream.
Here is a print screen from debugger showing this
[image]

To export the contents of the bitstream in a file, I use the following function
`    
PROCEDURE WRITE_STR_TO_FILE(Strm : adaasn1rtl.BitArray; 
			        DataLen	: Natural;
                                fileName:String) IS
      package Seq_IO is new Ada.Sequential_IO(adaasn1rtl.Asn1Byte);
      use Seq_IO;

      MASKS : CONSTANT adaasn1rtl.OctetBuffer (1 .. 8) := (16#80#, 16#40#, 16#20#, 16#10#, 16#08#, 16#04#, 16#02#, 16#01#);
      curByte:adaasn1rtl.Asn1Byte:=0;
      bitIndex:Integer;
      My_Out_File : Seq_IO.FILE_TYPE;
    BEGIN
      Create(My_Out_File, Out_File, fileName);

      FOR I IN 1..DataLen LOOP
         bitIndex := ((I-1) mod 8) +1;
         IF bitIndex = 1 THEN
            curByte:=0;
         END IF;
         IF Strm(I)=1 THEN
           curByte := curByte OR MASKS(bitIndex);
         END IF;
         IF bitIndex = 8 OR I = DataLen THEN
            Write(My_Out_File, curByte);
         END IF;
      END LOOP;

      Close(My_Out_File);
    END;    
`
Using this function, the contents will be written the way you expect.

gregerts, 2018-05-18:
Thank you for looking into my problem. Does this mean that a memory buffer that was encoded with the C library cannot be directly decoded by the Ada library and vica versa? And do I need to reverse the bits of incoming IMU messages (at 200 Hz) if I want to decode with Ada?

gmamais, 2018-05-19:
I am afraid so. The runtime libraries of C and Ada were developed at different times using a different strategy. The C RTL has a byte stream and is much easier to dump it to a file. Ada uses a BitArray which at the time of writing was considered to be a more clear and more understandable way to deal with bits encoding.

maxime-esa, 2018-05-19:
Just to rephrase it and make sure the original concern is clearly addressed: the ACN encoding is identical between C and Ada and there is no need to reverse the bits to decode it in Ada if encoded in C.

maxime-esa, 2018-05-21:
OK, after some further tests it seems indeed that there is an issue with one of the two runtimes. We have to investigate further and provide a solution to this issue.

gregerts, 2018-05-22:
Thank you, this is what I would expect. If I encode a datagram with the C library I should be able to decode it with the SPARK version without altering the bits first. As stated in my initial comment, the problem seems to be the usage of packed bit-arrays in Ada, and that your expected bit-order is not the one produced by GNAT that sees index 0 as the least significant of the byte. I would recommend to alter the SPARK library to work on byte-arrays (perhaps even from the standard package `System.Storage_Elements`) and follow the pattern as in the C library. I tested the encoding the message above one million times, and the C library is 16 times faster than Ada with the same compiler optimizations, so there also seems to be a significant performance hit on using bits instead of bytes. I am willing to contribute on this work.

maxime-esa, 2018-05-24:
@gmamais :Can this we worked in isolation, i.e. only by updating the RTL (adaasn1rtl.ad?) and perhaps the string template outputs for Spark backend, or would this imply changes in the code of the compiler?

gmamais, 2018-05-24:
Yes, the changes can be implemented in isolation (files C:\prj\GitHub\asn1scc\SPARK_RTL\adaasn1rtl.adb and C:\prj\GitHub\asn1scc\SPARK_RTL\adaasn1rtl.ads) . 

The Ada functions that must be modified are only the ones that actually read or write directly in the BitArray. These functions are:
BitStream_AppendBit, BitStream_ReadBit, BitStream_AppendByte, BitStream_DecodeByte
BitStream_AppendPartialByte, BitStream_ReadNibble, BitStream_Encode_Non_Negative_Integer
BitStream_Decode_Non_Negative_Integer, Acn_Enc_Boolean_true_pattern, Acn_Dec_Boolean_true_pattern, Acn_Enc_Boolean_false_pattern, Acn_Dec_Boolean_false_pattern
Acn_Enc_NullType_pattern, Acn_Dec_NullType_pattern, Acn_Enc_NullType_pattern2, 
Acn_Dec_NullType_pattern2

Moreover the definition of the BitArray must change from
type BitArray is array (Natural range <>) of BIT;
to 
type BitArray is array (Natural range <>) of Asn1Byte; 

if the changes to these specific functions are restricted only inside their body (and not in their interface) then no modification is required in the code of the ASN.1 compiler.

maxime-esa, 2018-05-24:
Thanks. Before optimizing the Ada runtime, perhaps we should first establish clearly who is currently right between `C` and `Ada` about the bit ordering. The `uPER` standard should be the reference - since we also need to check interoperability with other `ASN.1` tools. Have you checked already?

gmamais, 2018-05-26:
The bit order of C is definitely right. In the past, we used to compare the generate bitstream against another compiler at the bit level and the encoded buffer was written to the file as is without any bit reordering. On the other hand, for the Ada comparison against the other ASN.1 compiler, we had first to invoke the WRITE_STR_TO_FILE procedure which internally changes the order of the bits in each byte.

maxime-esa, 2019-03-19:
The issue has been fixed in the master branch, now Ada and C encodings are fully aligned.

usr3-1415, 2019-12-19:
Issue has been fixed. Ada and C encodings are fully aligned.

closed 2019-12-19 by usr3-1415