esa/asn1scc #157: [V4] --version crashes the application

Lurkerpas, 2018-07-03:
```
Asn1f4.exe --version

Wyjątek nieobsłużony: System.NullReferenceException: Odwołanie do obiektu nie zostało ustawione na wystąpienie obiektu.
   w Antlr.VersionInformation.printGitVersion() w C:\ASN1\asn1scc-ASN1SCC_V4_2018_07_02\asn1scc-ASN1SCC_V4\Antlr\VersionInformation.cs:wiersz 15
   w Program.main0(String[] argv) w C:\ASN1\asn1scc-ASN1SCC_V4_2018_07_02\asn1scc-ASN1SCC_V4\Asn1f4\Program.fs:wiersz 344
   w Program.main(String[] argv) w C:\ASN1\asn1scc-ASN1SCC_V4_2018_07_02\asn1scc-ASN1SCC_V4\Asn1f4\Program.fs:wiersz 367
```

```
Asn1f4.exe -v

Wyjątek nieobsłużony: System.NullReferenceException: Odwołanie do obiektu nie zostało ustawione na wystąpienie obiektu.
   w Antlr.VersionInformation.printGitVersion() w C:\ASN1\asn1scc-ASN1SCC_V4_2018_07_02\asn1scc-ASN1SCC_V4\Antlr\VersionInformation.cs:wiersz 15
   w Program.main0(String[] argv) w C:\ASN1\asn1scc-ASN1SCC_V4_2018_07_02\asn1scc-ASN1SCC_V4\Asn1f4\Program.fs:wiersz 344
   w Program.main(String[] argv) w C:\ASN1\asn1scc-ASN1SCC_V4_2018_07_02\asn1scc-ASN1SCC_V4\Asn1f4\Program.fs:wiersz 367
```

hcorg, 2018-07-04:
my guess: error appears when asn1scc was build from exported source, not from repo (for example - when downloading ZIP from gitthub).

imho it's another argument for removing from build process step which embeds git version information in project .... (other being: 1. it's not useful for the user, as this "number" is not sequential, 2. lib used for it does not work nicely on Linux and forces to introduce Makefile, which is strange for mono-based project, 3. it could be easily replaced by 'build number' tracked by CI...)

maxime-esa, 2018-07-04:
I agree that this version system is not optimal and I am also in favor of removing it, especially since it does not follow the actual version number (it stays stuck at tag 3.3.10)

However I do support the Makefile, which makes the build process consistent with all other applications.

hcorg, 2018-07-04:
other in TASTE? but for asn1scc could be used as stand-alone tool ;) also: Makefile has no sense on Windows, and .NET based application should do everything in msbuild/xbuild, just not to confuse users and developers. Makefile can exist for compatibility, but it build step should contain only xbuild command.

maxime-esa, 2018-07-04:
I don't mean TASTE, I mean nearly all Linux applications are built with Makefiles - whatever this makefile does, including calling language-specific build scheme (xbuild, cmake, qmake, gprbuild, ant, whatever). When you build from source, you should not need to know all the details of the ecosystem linked with the implementation language of the tool ; if anything else, for pragmatic reasons...

Of course, running the actual language-dependent build tool directly should _also work_ ! (and I think it does here, since the tool is developed under Windows with VS)

hcorg, 2018-07-04:
fortunately more and more application are switching from Makefile to more reasonable build-tools, but ok, I agree that for Linux (or Debian to be more precise) users Makefile should be there. But still - it should not do anything else than call nuget&xbuild (for example - for sake of those who would like to use MonoDevelop etc., or people like me, who, when facing .sln file, just call `nuget restore && xbuild` and hope for the best ;) )

usr3-1415, 2018-09-16:
Issue fixed by removing the dependency of the ASN.1 compiler from the git version system

closed 2018-09-16 by usr3-1415