esa/asn1scc #160: SEQUENCE OF or OCTET STRING etc has no constraints

TrymoCore, 2018-07-13:
I have a Problem with the ASN1SCC-Compiler. I want to compile some syntax-description to transfer bytes and byte arrays.
Every time when I use a SEQUENCE OF, I get the error from the headline. I've tried the same asn with OSS-Tools and it worked perfectly.

My easy example "EasyTest.asn":
`MoviLinkASNOneMaster DEFINITIONS AUTOMATIC TAGS ::=`
`BEGIN`
`ByteArray ::= SEQUENCE (SIZE (1..200)) OF OCTET STRING`
`END`

Output:
------------------------------------------------------------------------------------------------
asn1scc-bin-3.3.9\asn1.exe -c -BER EasyTest.asn
*****************************************************************************
* A fatal error has occurred
* Unexpected internal compiler error
*
*
* Please report bugs to: https://github.com/ttsiodras/asn1scc/issues
* Error is:
* SEQUENCE OF or OCTET STRING etc has no constraints
*
* Call stack is:
*
   bei Microsoft.FSharp.Core.Operators.Raise[T](Exception exn)
   bei c_h.SizeableTypeUperRange@28.Invoke(Unit unitVar0)
   bei c_h.PrintTypeDeclaration(Asn1Type t, FSharpList`1 p, AstRoot r)
   bei c_h.PrintTypeAss(TypeAssignment t, Asn1Module m, Asn1File f, AstRoot r, AcnAstResolved acn, Int32 curErrCode)
   bei c_h.PrintFile@226.Invoke(Int32 s, Tuple`2 tupledArg)
   bei CloneTree.foldMap[a,b,c](FSharpFunc`2 func, a state, FSharpList`1 lst)
   bei c_h.PrintFile(Asn1File f, String outDir, String newFileExt, AstRoot r, AcnAstResolved acn)
   bei c_h.DoWork@265.Invoke(Asn1File f)
   bei Microsoft.FSharp.Collections.SeqModule.Iterate[T](FSharpFunc`2 action, IEnumerable`1 source)
   bei c_h.DoWork(AstRoot r, AcnAstResolved acn, String outDir, String newFileExt)
   bei c_body.DoWork(AstRoot r, AcnAstResolved acn, String outDir)
   bei Asn1f2.Program.CheckSuccess(IEnumerable`1 args)
   bei Asn1f2.Program.Main2(String[] args)
   bei Asn1f2.Program.Main(String[] args)
*****************************************************************************
------------------------------------------------------------------------------------------------
Am I doing something wrong?!

TrymoCore, 2018-07-13:
Thanks for your fast reply.
I've tried the newest 4.1b binary version too, but it alwasy crashes instantly after starting the .exe. Even without any parameters. It tells that "FSharp.Core" is missing (but the file is present). Do I need some other librarys to execute 4.1b correctly that where not needed for 3.3.9? I will try building it manually with Visual Studio soon if there's no better solution...

Was there an support for BER in the past? I red the Chapter 10 of the TASTE manual where BER is desribed as a valid parameter. Is there an up-to-date documentation of the compiler?

maxime-esa, 2018-07-13:
Regarding the crash of the beta release: perhaps this was not tested on Windows before - sorry for that.  There is probably a DLL mismatch with the version installed on your system. We will look into that...

If you build it from source, I suppose this will work (the main development is made on Windows, so that should work from Visual Studio). In addition you'll get all the latest fixes.

Regarding BER there used to be a protoype but we found out that this was out of scope for space applications - that usually work on low resources and can't afford these kind of encodings. We also try to limit the features in order to save our own resources for maintenance. 

In the past some external contributors proposed to develop BER support (see here : https://github.com/ttsiodras/asn1scc/issues/12) but I did not hear about it afterwards, so I assume this idea was dropped.

closed 2018-07-15 by maxime-esa