esa/asn1scc #162: [V4] [AST XML] ENUMERATED is missing `size` and `encoding` properties in XML

hcorg, 2018-08-03:


gmamais, 2018-09-01:
The issue cannot be reproduced.
With the following grammars:
ASN.1
`
MyModel DEFINITIONS ::= BEGIN

MyEnum ::= ENUMERATED { i1 (10), i2 (20)}

END
`
ACN
`
MyModel DEFINITIONS ::= BEGIN

MyEnum [encoding pos-int, size 16, encode-values]

END
`
The generated AST xml contains both encoding and size property
`
          <TypeAssignment Name="MyEnum" CName="MyEnum" AdaName="MyEnum" Line="3" CharPositionInLine="0">
            <Asn1Type id="MyModel.MyEnum" Line="3" CharPositionInLine="11" ParameterizedTypeInstance="false">
              <ENUMERATED acnMaxSizeInBits="16" acnMinSizeInBits="16" uperMaxSizeInBits="1" uperMinSizeInBits="1" size="16" encoding="pos-int" encode-values="true">
                <Items>
                  <Item Name="i1" CName="i1" AdaName="i1" Value="10" Line="3" acnEncodeValue="10" CharPositionInLine="24" />
                  <Item Name="i2" CName="i2" AdaName="i2" Value="20" Line="3" acnEncodeValue="20" CharPositionInLine="33" />
                </Items>
                <Constraints>
                  <OR>
                    <EnumValue>i1</EnumValue>
                    <EnumValue>i2</EnumValue>
                  </OR>
                </Constraints>
                <WithComponentConstraints />
              </ENUMERATED>
            </Asn1Type>
          </TypeAssignment>

`

hcorg, 2018-09-03:
true, no longer reproducible

closed 2018-09-03 by hcorg