esa/asn1scc #163: [V4] [AST XML] ENUMERATED `encode-values` attribute contains always `false`

hcorg, 2018-08-03:


gmamais, 2018-09-01:
I can't reproduce this issue. I create the following grammars
ASN1
`MyModel DEFINITIONS ::= BEGIN

MyEnum ::= ENUMERATED { i1 (10), i2 (20)}

END
`
and ACN
`MyModel DEFINITIONS ::= BEGIN

MyEnum [encode-values]

END`

and the generated XML is:
`          <TypeAssignment Name="MyEnum" CName="MyEnum" AdaName="MyEnum" Line="3" CharPositionInLine="0">
            <Asn1Type id="MyModel.MyEnum" Line="3" CharPositionInLine="11" ParameterizedTypeInstance="false">
              <ENUMERATED acnMaxSizeInBits="4" acnMinSizeInBits="4" uperMaxSizeInBits="1" uperMinSizeInBits="1" encode-values="true">
                <Items>
                  <Item Name="i1" CName="i1" AdaName="i1" Value="10" Line="3" acnEncodeValue="10" CharPositionInLine="24" />
                  <Item Name="i2" CName="i2" AdaName="i2" Value="20" Line="3" acnEncodeValue="20" CharPositionInLine="33" />
                </Items>
                <Constraints>
                  <OR>
                    <EnumValue>i1</EnumValue>
                    <EnumValue>i2</EnumValue>
                  </OR>
                </Constraints>
                <WithComponentConstraints />
              </ENUMERATED>
            </Asn1Type>
          </TypeAssignment>
`

where as you can see the attribute encode-values is true.

hcorg, 2018-09-03:
indeed seems it got fixed in the mean time :)

closed 2018-09-03 by hcorg