esa/asn1scc #166: [V4] [AST XML] Include Parametrized Types information in XML

hcorg, 2018-08-08:
Should contain also parametrized argument list with possible constraints.

Also - instances should reference which parametrized type they instantiate and list used parameters.

maxime-esa, 2021-01-15:
I think this is now implemented

Given this parameterized type:

```
Foo DEFINITIONS ::=
BEGIN
 Seq {Foo, INTEGER:end-point} ::= SEQUENCE {
   innerChoice CHOICE {
     option1 Foo,
     option2 INTEGER (0..end-point)
   }
 }

  Bar ::= BOOLEAN
  Instance ::= Seq {Bar, 5}
END
```

If you generate the complete XML dump of the AST like this:
```
asn1.exe -x full.xml foo.asn
```

There are some references to the parameterized arguements (types only), for example:
```
  <Asn1Type id="Foo.Instance" ParameterizedTypeInstance="true">
```
and
```
<REFERENCE_TYPE Module="Foo" TypeAssignment="Bar">
  <Asn1Type id="Foo.Instance.innerChoice.option1" tasInfoModule="Foo" tasInfoName="Bar">
 <Constraints />
 <WithComponentConstraints />
.....
 </Asn1Type>
</REFERENCE_TYPE>
```

Do you need more?

maxime-esa, 2024-03-07:
@hcorg if anything is missing feel free to reopen the issue