esa/asn1scc #167: Message_REQUIRED_BITS_FOR_ENCODE is wrong for CHOICE elements

stevew1154, 2018-08-11:
In the generated code for a CHOICE, the #define for the Message_REQUIRED_BITS_FOR_ENCODE seems to be incorrect. For example, in a CHOICE with 4 choices, the encode/decode functions themselves correctly will only produce or consume 2 bits for the choice selector, but the constant will indicate 3 bits are needed, and since that same constant is used internally to generate the Message_REQUIRED_BYTES_FOR_ENCODE, this error can propagate to cause users to allocate more memory than required for their messages.

In the attached choices.asn.txt, the types ChoiceOf1, ... ChoiceOf4 generated:

```
#define ChoiceOf1_REQUIRED_BITS_FOR_ENCODING        8
#define ChoiceOf1_REQUIRED_BITS_FOR_ACN_ENCODING    8
#define ChoiceOf2_REQUIRED_BITS_FOR_ENCODING        9
#define ChoiceOf2_REQUIRED_BITS_FOR_ACN_ENCODING    9
#define ChoiceOf3_REQUIRED_BITS_FOR_ENCODING        9
#define ChoiceOf3_REQUIRED_BITS_FOR_ACN_ENCODING    9
#define ChoiceOf4_REQUIRED_BITS_FOR_ENCODING        10
#define ChoiceOf4_REQUIRED_BITS_FOR_ACN_ENCODING    10

```
The correct values should be 7, 8, 9, and 9.

[choices.asn.txt](https://github.com/ttsiodras/asn1scc/files/2279792/choices.asn.txt)

maxime-esa, 2018-08-11:
This bug is solved in branch `ASN1SCC_V4`.

It's a major refactoring of the tool fixing most bugs from V3 and introducing a few new features (mostly ACN-related). Strongly recommended, even if still at beta state.

If you build the tool yourself and you work on Linux, just do:
```
$ git clone https://github.com/ttsiodras/asn1scc asn1scc_v4
$ cd asn1scc_v4 && git checkout ASN1SCC_V4
$ make
```
and the result will be in the folder `Asn1v4/bin/Debug`

I tried it with your grammar and it works as expected

stevew1154, 2018-08-11:
Maxime,

Thank you for your quick reply. That does seem to fix it, but:

1) I can't use the ASN1SCC_V4 on my ubuntu 14.04 machine because the version of mono is too old.

2) I can use ASN1SCC_V4 on my ubuntu 18.04 machine, but I had to work around a problem with the "sudo apt-get install -y --force-yes libgit2-24 libgit2-dev libssl-dev" in  patchGitVersionTask.sh wasn't working correctly. On ubuntu-18.04, we are at libgit2-26.

I was somewhat surprised by how much the generated code is re-arranged.

Also, the Asn1fv4 tool seems to always generate the Xxxxx_Equal() functions now, as if the "--equal-func" argument is always turned on. Can that be turned off?

stevew1154, 2018-08-16:
Maxime,

It works just to remove the "libgit2-24". The "libgit2-dev" depends on the right version.

Steve

gmamais, 2018-09-01:
The latest commit fixes the issue of always generating  the _Equal functions.
Please note that if you generated automatic test cases (-atc) the _Equal functions will be generated even if you omit the -equal argument. Execution of automatic test cases requires the presence of _Equal functions in order to compare the decoded and encoded values.

closed 2018-09-01 by gmamais