esa/asn1scc #168: UPER size for "BIT STRING (SIZE (N))" is 8 times too large.

stevew1154, 2018-08-22:
Using ASN1SCC_V4 branch:

```
Foo ::= SEQUENCE {
   spare                          BIT STRING (SIZE (68))
}
```

Yields a reasonable structure Foo:
```
/*-- Foo --------------------------------------------*/
typedef struct {
    
    byte arr[9];
} Foo_spare;

typedef struct {
    Foo_spare spare;

} Foo;
```
And the decode/encode functions properly consume/produce 68 bits, but the sizing constants are off by a factor of 8:

```
#define Foo_REQUIRED_BYTES_FOR_ENCODING       68 
#define Foo_REQUIRED_BITS_FOR_ENCODING        544
```

stevew1154, 2018-08-23:
My git commit is:

97dbd82 mantis 0000663 (almost completed)

gmamais, 2018-09-01:
Thanks for reporting this bug.
The bug, which also affected the ICD generation, was fixed.

closed 2018-09-01 by gmamais