esa/asn1scc #172: asn1scc crashes on branch ASN1_V4

AlexanderFabisch, 2018-09-07:
After building successfully with #171, I tried to run asn1scc:

```
$ ./Asn1f4.exe 
Can't find custom attr constructor image: /tmp/asn1scc/Asn1f4/bin/Debug/Asn1f4.exe mtoken: 0x0a00006e due to: Could not load file or assembly 'FSharp.Core, Version=4.4.1.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
Can't find custom attr constructor image: /tmp/asn1scc/Asn1f4/bin/Debug/Asn1f4.exe mtoken: 0x0a000001 due to: Could not load file or assembly 'FSharp.Core, Version=4.4.1.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
Can't find custom attr constructor image: /tmp/asn1scc/Asn1f4/bin/Debug/Asn1f4.exe mtoken: 0x0a00006e due to: Could not load file or assembly 'FSharp.Core, Version=4.4.1.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.

Unhandled Exception:
Nested exception detected.
Original Exception: at Program.main (string[]) [0x00000] in <5b9273ba7179049aa7450383ba73925b>:0

Nested exception:at (wrapper managed-to-native) System.MonoCustomAttrs.IsDefinedInternal (System.Reflection.ICustomAttributeProvider,System.Type) [0x00049] in <2943701620b54f86b436d3ffad010412>:0
at System.MonoCustomAttrs.IsDefined (System.Reflection.ICustomAttributeProvider,System.Type,bool) [0x00027] in <2943701620b54f86b436d3ffad010412>:0
at System.Reflection.MonoMethod.IsDefined (System.Type,bool) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Attribute.IsDefined (System.Reflection.MemberInfo,System.Type,bool) [0x00088] in <2943701620b54f86b436d3ffad010412>:0
at System.Attribute.IsDefined (System.Reflection.MemberInfo,System.Type) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Reflection.CustomAttributeExtensions.IsDefined (System.Reflection.MemberInfo,System.Type) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.GetFullNameForStackTrace (System.Text.StringBuilder,System.Reflection.MethodBase,bool,bool&) [0x0004f] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.AddFrames (System.Text.StringBuilder) [0x00080] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.ToString () [0x00052] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.ToString (System.Diagnostics.StackTrace/TraceFormat) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Environment.GetStackTrace (System.Exception,bool) [0x00014] in <2943701620b54f86b436d3ffad010412>:0
at System.Exception.GetStackTrace (bool) [0x00038] in <2943701620b54f86b436d3ffad010412>:0
at System.Exception.get_StackTrace () [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.IO.FileNotFoundException.ToString () [0x00075] in <2943701620b54f86b436d3ffad010412>:0

Can't find custom attr constructor image: /tmp/asn1scc/Asn1f4/bin/Debug/Asn1f4.exe mtoken: 0x0a00006e due to: Could not load file or assembly 'FSharp.Core, Version=4.4.1.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
[ERROR] FATAL UNHANDLED EXCEPTION: Nested exception detected.
Original Exception: at Program.main (string[]) [0x00000] in <5b9273ba7179049aa7450383ba73925b>:0

Nested exception:at (wrapper managed-to-native) System.MonoCustomAttrs.IsDefinedInternal (System.Reflection.ICustomAttributeProvider,System.Type) [0x00049] in <2943701620b54f86b436d3ffad010412>:0
at System.MonoCustomAttrs.IsDefined (System.Reflection.ICustomAttributeProvider,System.Type,bool) [0x00027] in <2943701620b54f86b436d3ffad010412>:0
at System.Reflection.MonoMethod.IsDefined (System.Type,bool) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Attribute.IsDefined (System.Reflection.MemberInfo,System.Type,bool) [0x00088] in <2943701620b54f86b436d3ffad010412>:0
at System.Attribute.IsDefined (System.Reflection.MemberInfo,System.Type) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Reflection.CustomAttributeExtensions.IsDefined (System.Reflection.MemberInfo,System.Type) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.GetFullNameForStackTrace (System.Text.StringBuilder,System.Reflection.MethodBase,bool,bool&) [0x0004f] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.AddFrames (System.Text.StringBuilder) [0x00080] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.ToString () [0x00052] in <2943701620b54f86b436d3ffad010412>:0
at System.Diagnostics.StackTrace.ToString (System.Diagnostics.StackTrace/TraceFormat) [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.Environment.GetStackTrace (System.Exception,bool) [0x00014] in <2943701620b54f86b436d3ffad010412>:0
at System.Exception.GetStackTrace (bool) [0x00038] in <2943701620b54f86b436d3ffad010412>:0
at System.Exception.get_StackTrace () [0x00000] in <2943701620b54f86b436d3ffad010412>:0
at System.IO.FileNotFoundException.ToString () [0x00075] in <2943701620b54f86b436d3ffad010412>:0
```

AlexanderFabisch, 2018-09-07:
yes, seems to work

maxime-esa, 2018-09-07:
Ok, so it seems that the dll of the FSharp Core library that ends up in Asn1f4/bin/Debug is not right. If you list side by side all the FSharp*.dll from the binary release and the one you had installed, the size probably differ?
Mono/Windows DLL mess on Ubuntu goes beyond my knowledge - I let others help with that :-)

(We will make a new binay release in a short while with all the latest commits, there were a few fixes since the one you downloaded)

AlexanderFabisch, 2018-09-07:
> If you list side by side all the FSharp*.dll from the binary release and the one you had installed, the size probably differ?

Could be possible. I don't know where those dlls are located.

> (We will make a new binay release in a short while with all the latest commits, there were a few fixes since the one you downloaded)

I'm wating for it. Unfortunately there are some breaking changes in the generated code in comparison to the compiler that we used previously, for example, `<typename>_REQUIRED_BYTES_FOR_ACN_ENCODING` is not generated for each type.

maxime-esa, 2018-09-07:
Our plan is that V4 should be nearly entirely backward-compatible with the V3 ; we are working on the missing bits before fully replacing V3 in the master branch (i.e. moving out of beta state), and if you find some incompatibilities like this, we will fix them (please report them as separate tickets) - or provide justification and workaround.

closed 2018-09-16 by maxime-esa