esa/asn1scc #175: XSD for the AST XML dump

andrestt, 2018-09-20:
Hello!

Does there exist an XSD schema and/or some other specification for the XML dump generated by the -x option? We are going to read the AST dump generated by ASN1SCC in a Java tool and a specification of the format would be useful.

Thanks in advance,
Andres Toom

IB Krates / AdaCore

maxime-esa, 2018-09-20:
There isn't. Note that there are two forms of the AST, the complete one you get with -x is exhaustive, it contains all ACN information in addition to the types with extensive constraints.

There is also a lighter AST which is also customisable at will, and this is the one we use in TASTE. 
It is based on a template file named `xml.stg` (present in the binary folder of the compiler).
if you run:

```
asn1.exe -customStg /path/to/xml.stg:ast.xml -customStgAstVersion 4 <asn1 files>
```

You'll get the AST and you'll see the structure is straightforward. 
If you edit `xml.stg` you'll be able to understand how it is created. The file python.stg is based on the same file and creates the AST that is used by `Opengeode`.

andrestt, 2018-09-20:
OK. Thanks for these tips!

Andres

closed 2018-09-20 by maxime-esa