esa/asn1scc #176: Does not build on 64-bit only Mac

cjheath, 2018-12-10:
./build.sh says

```
error : 'OutputPath' property is not set for this project. Usually this is caused by invalid Configuration/Platform combination. Original values: Configuration: Debug Platform: x86.
```

... etc. It needs to have a 64-bit OutputPath setting.

maxime-esa, 2018-12-10:
Try only running `make` or `make -f Makefile.Release` instead of `build.sh`. It is an old script that was used for building older versions.

cjheath, 2018-12-10:
"make" makes a lot of progress, but fails when it says
```
cp -al Asn1f4/bin/Debug/Asn1f4.exe Asn1f4/bin/Debug/asn1.exe 
cp: illegal option -- l
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
```

It looks like that is the final step before completion. Thank you. Please update README.md, and perhaps Makefile?

ttsiodras, 2018-12-10:
Updated both of them in ASN1SCCv4 branch (which will be merged in master).

closed 2018-12-10 by ttsiodras

maxime-esa, 2018-12-10:
`master` is up to date, don't use `ASN1SCC_V4` branch anymore - it is behind.

ttsiodras, 2018-12-10:
Yep - and merge already completed.