esa/asn1scc #182: Buil failure on ubuntu 18.10

osmoc, 2019-03-18:
I've installed dependencies according to readme and got following error after 'make' on ubuntu 18.10 x86_64:
`
nuget restore

Unhandled Exception:
System.TypeInitializationException: The type initializer for 'System.Console' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.ConsoleDriver' threw an exception. ---> System.Exception: Magic number is wrong: 542
  at System.TermInfoReader.ReadHeader (System.Byte[] buffer, System.Int32& position) [0x0002b] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.TermInfoReader..ctor (System.String term, System.String filename) [0x00065] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.TermInfoDriver..ctor (System.String term) [0x00058] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.ConsoleDriver.CreateTermInfoDriver (System.String term) [0x00000] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.ConsoleDriver..cctor () [0x00062] in <8f2c484307284b51944a1a13a14c0266>:0 
   --- End of inner exception stack trace ---
  at System.Console.SetupStreams (System.Text.Encoding inputEncoding, System.Text.Encoding outputEncoding) [0x0000a] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.Console..cctor () [0x000a8] in <8f2c484307284b51944a1a13a14c0266>:0 
   --- End of inner exception stack trace ---
  at NuGet.Program.Main (System.String[] args) [0x00005] in <dc8a236afba24badbb185682f8956bec>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'System.Console' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.ConsoleDriver' threw an exception. ---> System.Exception: Magic number is wrong: 542
  at System.TermInfoReader.ReadHeader (System.Byte[] buffer, System.Int32& position) [0x0002b] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.TermInfoReader..ctor (System.String term, System.String filename) [0x00065] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.TermInfoDriver..ctor (System.String term) [0x00058] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.ConsoleDriver.CreateTermInfoDriver (System.String term) [0x00000] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.ConsoleDriver..cctor () [0x00062] in <8f2c484307284b51944a1a13a14c0266>:0 
   --- End of inner exception stack trace ---
  at System.Console.SetupStreams (System.Text.Encoding inputEncoding, System.Text.Encoding outputEncoding) [0x0000a] in <8f2c484307284b51944a1a13a14c0266>:0 
  at System.Console..cctor () [0x000a8] in <8f2c484307284b51944a1a13a14c0266>:0 
   --- End of inner exception stack trace ---
  at NuGet.Program.Main (System.String[] args) [0x00005] in <dc8a236afba24badbb185682f8956bec>:0 
`
I'm not sure how to interpret this - am I still missing some dependency?
The build was attempted on latest master: c440eb56f17d93f69c128326420bc5d9c99342ea

osmoc, 2019-03-18:
Seems like my nuget install is simply broken - I've got the same error anywhere. Any ideas what could be causing it? Have not used nuget before.

maxime-esa, 2019-03-18:
This is an issue with `nuget` or `mono`, there are many bug reports about this in other repositories using nuget (e.g. here https://github.com/KSP-CKAN/CKAN/issues/2611)

One thing you can try is to re-install mono, or try the command (`nuget restore`) from an `xterm` instead of the default terminal from Ubuntu.

maxime-esa, 2019-03-18:
It seems related to that bug report in mono: https://github.com/mono/mono/issues/6752

osmoc, 2019-03-18:
Got it, thanks for quick help. Indeed, the problem seems to be terminal-related: it's broken under screen but works fine in plain urxvt for example.

closed 2019-03-18 by osmoc