esa/asn1scc #183: RFE: package for major GNU/Linux distibutions

osmoc, 2019-03-19:
The process of building asn1scc seems non-trivial - would be great to make it more accessible for users by adding pre-built .deb (and perhaps .rpm) packages by utilizing PPA (https://help.launchpad.net/Packaging/PPA) and COPR (https://copr.fedorainfracloud.org/coprs/) infrastructure or cross-distro builder (https://build.opensuse.org/).

This would help to increase user adoption and will pave the way to eventually including asn1scc into distribution repositories by default.

maxime-esa, 2019-03-19:
We have pre-built releases already here: https://github.com/ttsiodras/asn1scc/releases

osmoc, 2019-03-19:
That's nice but still not as convenient as a native packages: no automatic update and dependency management for example.

maxime-esa, 2019-03-19:
It is not something we can do at the moment due to limited resources.. Are you interested to take care of this? Contributions are very welcome :-)

osmoc, 2019-03-19:
I wish I could :(
Let's keep it open to increase the chances that somebody with better mono knowledge than me or more time at hands than you would stumble upon this and volunteer to help :)

closed 2022-02-18 by ttsiodras