esa/asn1scc #186: String length not initialized

gregerts, 2019-05-02:
I am testing the master branch after having used an older release, and notice that the nCount of string does not seem to be initialized any more.

The type:

```
maxSize-T-String T-UInt32 ::= 256

T-String ::= OCTET STRING (SIZE(0..maxSize-T-String))
```
Produces the following initialization code:
```
void T_String_Initialize(T_String* pVal)
{
	memset(pVal->arr, 0x0, 256);
}
```
This fails to encode if the nCount is not explicitly set to 0. Is this the intended behaviour?

maxime-esa, 2019-05-02:
Confirmed, I also checked in Ada, it does set the value of Length but it sets it to 256 instead of 0

[image]

usr3-1415, 2019-05-02:
Issue fixed in the latest commit (version 4.1.0.11f)

closed 2019-05-02 by usr3-1415