esa/asn1scc #188: Lack of a comprehensive example

fischermario, 2019-05-11:
While searching for a proper and reliable ASN.1 implementation for encoding/decoding I came across your [blog post](http://ttsiodras.github.io/asn1.html). I tried to reproduce the example, but I could not figure out how to use the C code that is being produced by asn1.exe.

I have tried to use this repo's code as well as DMT (as suggested in the blog post). The resulting code of these two "approaches" differs, but when I try to encode and decode a message the outcome is the same.

At first I have created a file `sample.asn` with the following content:

```
MY-MODULE DEFINITIONS AUTOMATIC TAGS ::= BEGIN

Message ::= SEQUENCE {
    msgId INTEGER,
    myflag INTEGER,
    value REAL,
    szDescription OCTET STRING (SIZE(10)),
    isReady BOOLEAN
}

END
```

I have created the corresponding C sources by running:

`../asn1scc/Asn1f4/bin/Debug/asn1.exe -c -uPER sample.asn`

My demo program looks like the following (I really have no idea if this is even remotely correct):

```
#include <stdio.h>

#include "sample.h"

int main(int argc, char **argv) {
	Message msg;
	Message_Initialize(&msg);
	
	msg.msgId = (asn1SccSint)4;
	msg.myflag = (asn1SccSint)1;
	msg.value = (asn1Real)1.1;
	msg.myflag = (flag)true;
	
	byte encBuff[Message_REQUIRED_BYTES_FOR_ENCODING];
	BitStream bitStrm;
	int ErrCode = -1;
	
	BitStream_Init(&bitStrm, encBuff, Message_REQUIRED_BYTES_FOR_ENCODING);
	flag ret = Message_Encode(&msg, &bitStrm, &ErrCode, TRUE);
	
	printf("[Encode] Return flag: %d\n", (int)ret);
	printf("[Encode] Error code: %d\n", ErrCode);
	printf("[Encode] Buffer:\n");
	int idx = 0;
	for (idx = 0; idx < Message_REQUIRED_BYTES_FOR_ENCODING; idx++) {
		printf("%02X ", bitStrm.buf[idx]);
		if ((idx+1) % 10 == 0)
			printf("\n");
	}
	printf("\n");
	
	// -----------------------------------------------------------------
	
	Message dmsg;
	Message_Initialize(&dmsg);
	
	ret = Message_Decode(&dmsg, &bitStrm, &ErrCode);
	printf("[Decode] Return flag: %d\n", (int)ret);
	printf("[Decode] Error code: %d\n", ErrCode);
	printf("msgId: %d\n", (int)dmsg.msgId);
	printf("myflag: %d\n", (int)dmsg.myflag);
	printf("value: %.2f\n", (float)dmsg.value);
	printf("myflag: %d\n", (int)dmsg.myflag);
	
	return 0;
}
```

Unfortunately the output looks like this:

```
[Encode] Return flag: 1
[Encode] Error code: 0
[Encode] Buffer:
01 04 01 01 09 80 CC 11 99 99 
99 99 99 9A 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 
00 00 
[Decode] Return flag: 1
[Decode] Error code: 0
msgId: 0
myflag: 0
value: 0.00
myflag: 0
```

What am I missing? Why does the encoded buffer start with 01 04? I expected that the buffer would start with a sequence indicator and a length byte.

Thanks for your advice.

ttsiodras, 2019-05-12:
Hi Mario.

Away on a business trip (replying from my phone from an airport) - but a
very easy way to see the proper use of the generated APIs is to pass the
"-atc" option to ASN1SCC.

This will create standalone test code, including also a makefile - the
tests will initialize the structures, and check that after encoding and
decoding, the values survive the roundtrip. This should demonstrate
everything you need, specifically made for your ASN.1 grammar.

maxime-esa, 2019-05-12:
What do you think is wrong in that encoding?

`msgId` is an unconstrained integer. In uPER encodings this means that the actual value has to follow a length byte. Since you need only one byte to encode value "4", the first byte is 01 and the second is 04. That seems right to me.

usr3-1415, 2019-05-12:
Hi Mario,
Why do you expect this behavior? uPER (i.e. unaligned Packed Encoding Rules) doesn't indicators as BER (i.e. tags). Also, it uses a length determinant only in types with a size constraint such as SEQUENCE OF,  OCTET STRING, etc and when the size is not known at compile time.

maxime-esa, 2019-05-12:
@Mario, to make sure the encoding is right, you can also cross-check it with another encoder.

    $ pip install asn1tools
    $ python
    >>> import asn1tools                                                                                    
    >>> foo=asn1tools.compile_files('mario.asn', 'uper')                                                        
    >>> foo.decode ('Message', b'\x01\x04\x01\x01\x09\x80\xCC\x11\x99\x99\x99\x99\x99\x9A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')     
    {'msgId': 4, 'myflag': 1, 'szDescription': '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 'value': 1.1, 'isReady': False}

As you can see this is fully correct. In your test case you have a typo as you did not set `isReady` to true:

    msg.msgId = (asn1SccSint)4;
    msg.myflag = (asn1SccSint)1;
    msg.value = (asn1Real)1.1;
    msg.myflag = (flag)true;                  <-- typo

fischermario, 2019-05-12:
Thank you all very much for your support! Your hints have been a great help and I have learnt a lot in the last hours. Coming from Protobuf I now realize how little I knew about ASN.1 ;-)

I have updated my demo code and it now works as expected:

```
#include <stdio.h>

#include "sample.h"

int main(int argc, char **argv) {
	Message msg;
	
	msg.msgId = (asn1SccSint)4;
	msg.myflag = (asn1SccSint)1;
	msg.value = (asn1Real)1.1;
	msg.isReady = (flag)true;
	
	byte encBuff[Message_REQUIRED_BYTES_FOR_ENCODING + 1]; // +1 for zerosized types
	BitStream bitStrm;
	int ErrCode = -1;
	
	BitStream_Init(&bitStrm, encBuff, Message_REQUIRED_BYTES_FOR_ENCODING);
	flag ret = Message_Encode(&msg, &bitStrm, &ErrCode, TRUE);
	
	printf("[Encode] Return flag: %d\n", (int)ret);
	printf("[Encode] Error code: %d\n", ErrCode);
	
	if (ret) {
		printf("[Encode] Buffer:\n");
		int idx = 0;
		for (; idx < Message_REQUIRED_BYTES_FOR_ENCODING; idx++) {
			printf("%02X ", bitStrm.buf[idx]);
			if ((idx+1) % 10 == 0)
				printf("\n");
		}
		printf("\n");
		
		// -----------------------------------------------------------------
		
		Message dmsg;
		
		BitStream_AttachBuffer(&bitStrm, encBuff, Message_REQUIRED_BYTES_FOR_ENCODING);
		
		ret = Message_Decode(&dmsg, &bitStrm, &ErrCode);
		printf("[Decode] Return flag: %d\n", (int)ret);
		printf("[Decode] Error code: %d\n", ErrCode);
		
		if (ret) {
			printf("msgId: %d\n", (int)dmsg.msgId);
			printf("myflag: %d\n", (int)dmsg.myflag);
			printf("value: %.2f\n", (float)dmsg.value);
			printf("isReady: %d\n", (int)dmsg.isReady);
		}
	}
	
	return 0;
}
```

Actually it was the line

`BitStream_AttachBuffer(&bitStrm, encBuff, Message_REQUIRED_BYTES_FOR_ENCODING);`

which finally made the code work. I find that a little odd because when looking at the implementation of `BitStream_AttachBuffer` in `asn1crt.c` the function seems to already assign the buffer pointer to the BitStream struct's buf member. I thought that already happened with the call to `BitStream_Init`? Can somebody clarify this?

@maxime-esa: Shame on me for the typo. I have corrected my mistake in the code above.

@maxime-esa and @usr3-1415: As I said above I fell prey to some misconceptions regarding how uPER encoding works. I now know that the output is correct.

@all: Is there any ASN.1 encoding variant out there where the first byte indicates the length of the whole message? I am asking this because that would be very helpful for having a zero-copy decoder (when it comes to UART communication most of the usual suspects like TI and ST expect to know the number of bytes to read in advance). This is also one of the pain points with Protobuf.

Please excuse my rookie questions - I am really new to ASN.1 :-)

maxime-esa, 2019-05-13:
@fischermario 
yes, that's possible in ASN.1 using the CONTAINING keyword.

If the type of your payload is e.g `Payload ::= INTEGER (65535)`

Then you may create an enclosing type like this :
```
PDU ::= OCTET STRING (0..255) (CONTAINING Payload)
```

Because the octet string starts with a length field it will be added to the packet.

_Note however that our compiler does not support that feature at the moment._

We have a way to achieve this using `ACN`. `ACN` is a notation we created to complements the ASN.1 model with custom encodings rules.

These kind of length fields is part of a more general problem of adding information that is not directly available to the encoder at the time of the encoding (it must be added at the end, when all other fields are filled). Another case of that is when you want to add a checksum in your packet. The encoder cannot compute the value and place it before the whole packet is processed. 

This is how to proceed using ASN1SCC:

1. Create your payload as a SEQUENCE in your `.asn` file

```
PDU ::= SEQUENCE {
   payload BOOLEAN   -- whatever
}
``` 

And in a .acn file (you can get the skeleton if you call asn1scc with the `-ACND` flag) do:

```
PDU [post-encoding-function add-length] {
  length NULL [pattern '00000000'B, save-position],  -- add one byte at the begining of the packet
  payload []
}
```

This tells the compiler that a function "add-length" will have to be called by the binary encoder when it is done with the other fields. It will get as argument the position of the "length" field in the packet, allowing you to put the value at the right place with little effort.

You need to provide the implementation of "add-length". 

Please check the complete example here: https://github.com/ttsiodras/asn1scc/tree/master/mantis/0000824

You will find an ASN.1, ACN, C and Ada implementation of the add-length (or similar) function, and the command line you need to use to tell the compiler how to use these functions.

Let me know if that helped!

closed 2019-05-14 by ttsiodras