esa/asn1scc #189: asn1.exe failed loading FSharp.Core

pozzugno, 2019-06-03:
I downloaded asn1scc-bin-4.1.0.1f.tar.bz2 and tried to run asn1.exe in my Windows 10 command prompt, however it failed with an error regarding loading FSharp.Core Version=4.4.0.0 (the exact message is in italian).

I solved renaming asn1.exe in asn1f4.exe (I noticed the file asn1f4.exe.config). Is it the correct way to solve the problem?

pozzugno, 2019-06-03:
I'm sorry, I enjoyed too early. The problem doesn't disappear even after renaming the executable. The last working release for me is asn1scc-bin-3.3.22.tar.bz2.

maxime-esa, 2019-06-04:
Right, we noticed this. The release version works on Linux only, probably because it comes with fsharp DLLs that conflicts with the .NET version on Windows.
One solution on Windows would be to rebuild from source using the VS project file. Is that something you can do?

multiplemonomials, 2020-06-04:
I built the latest version for Windows if anyone needs it: https://app.box.com/shared/static/wjmfvtmxyabsv08c7gq07tamumv9hhup.zip

maxime-esa, 2020-06-04:
Thanks, well appreciated!

multiplemonomials, 2020-06-06:
Quick (unrelated) question: is there any way to specify the size of REALs with asn1scc?  I noticed the `--fp-word-size` compile option, but this seems to apply globally to all REALs -- it would be cool if there was a way to specify certain REALs as C doubles and others as floats.  I tried a file containing
```
Float32 ::= REAL (WITH COMPONENTS {
                    mantissa (-16777215..16777215),
                    base (2),
                    exponent (-149..104)
                })
```
but I got
```
ProgrammingTypes.asn:23:20: error: mismatched input 'mantissa' expecting LID
```

If this isn't currently supported, it would be a much appreciated feature at USC RPL.

Also, what's the recommended way to interoperate with other programs (python asn1tools) that have a standard-conforming REAL?  Is there some form of REAL type that will be processed equally by both systems?

maxime-esa, 2020-06-06:
Hi Jamie,

1. At the moment we have not considered this option to keep it simpler. We have the same question with integers - we only generate 64 bits integers (signed or unsigned). The range only influences the size of the uPER/ACN encoded data, not the native type. We would probably need more code (encoding/decoding functions for each type) and that would create overhead in the RTL.

2. We are looking into it. This form is currently not properly supported

3. There are different ways to encode real numbers in uPER and if I recall correctly not all tools support all the possible forms. One of the forms we don't support is text encoding (I think it is the form used in OSS Playground tool). ASN1SCC uses the binary form and should be decoded properly by the python asn1tools.
I just made a test, I encoded with asn1scc the value 255.0 and asked asn1tools to decode the resulting binary stream - which worked as expected:
[image]
I showed another example with a REAL field in the middle of a sequence and the decoding with asn1tools went well too: https://github.com/ttsiodras/asn1scc/issues/188

multiplemonomials, 2020-06-06:
Alright, that worked great!  A naked REAL type with no constraints works fine and is encoded and decoded correctly by asn1scc and asn1tools.  (though it isn't accepted by the DMT asn-to-python mapper... but asn1tools is likely a better solution for our situation)

I'm very glad I found asn1scc -- I've been experimenting with asn1c and it's... not ideal for embedded devices.  The method of preventing it from allocating memory is very complex (and undocumented), and its code needs a lot of post-generation patches to build on any system that isn't Linux.  It also just seems to generate a really excessive amount of code for what it does.  Asn1scc seems to be an order of magnitude more lightweight.

Last question, promise: is there anything (constraints or the value of the number) that determines the size of a REAL type in the encoded packet?  Or is it always represented as a IEEE double?  We don't really care about the C type but we are interested in making the encoded packet as small as possible.

maxime-esa, 2020-06-06:
Thanks ! Indeed we are trying to make optimal and readable code to target embedded applications. 

REAL types with constraints work as well, if you express the constraints like this:
```MyReal ::= REAL (-100.0 .. 100.0)```
and not with the mantissa syntax.

Regarding the size of REAL in encoded packets:
If you use ACN, you can control the size to be 32 (Float) or 64 bits (Double). Check the documentation here: https://taste.tuxfamily.org/wiki/index.php?title=Technical_topic:_ASN.1_-_An_introduction_to_ACN

If you use uPER then the encoding size depends on the contents of the real value.
In general, the uPER binary encoding is as follows:

```mantissa*base^Exponent```

where base can be either 2 8 or 16. 

The encoding bytes are as follows:

* one byte for the length of the encoding real value (values in bytes)
* one byte for the header (contains the base (2, 8, or 16), the length of the mantissa in bytes and the length of the expoent
* the mantissa
* the exponent
 
The length of mantissa and exponent depends on whether you use 32 or 64 bits Floating-point arithmetic and on the content the real value.

In real-world scenarios, the mantissa is up to 7 bytes and the exponent up to 3 bytes.
So, the answer to your question is that there is no way to affect the encoding size in `REAL` numbers.

If you  are interested in making the encoded packet as small as possible, then perhaps you should be considered using fixed-size integers.

closed 2021-01-15 by maxime-esa