esa/asn1scc #200: Support IMPORTS in corresponding ACN

redcactus, 2020-04-21:
I have two couples ASN/ACN files in [Types.zip](https://github.com/ttsiodras/asn1scc/files/4512214/Types.zip). One module have EXPORTS clause, another IMPORTS it. While trying to compile them like 
`asn1f4.exe -o . -ACN -c Types1.asn Types1.acn Types2.asn Types2.acn`
I see error
`Types2.acn:9:9: error: No type assignment defined with name 'IUINT8' in module 'Types2'`
Whether I erase `Types2.acn` from command line result is successfull, but I have no control of array size in thas case.
Is in possilbe to use imported types in ACN files?

maxime-esa, 2020-04-21:
You can, but you have to reference the full path:

Change Types2.acn to :

   Data<**Types1.IUINT8**:nElements> []

redcactus, 2020-04-23:
Thanks a lot!
Is there any document where full ACN syntax is described?

maxime-esa, 2020-04-24:
It is documented here: https://taste.tuxfamily.org/wiki/index.php?title=Technical_topic:_ASN.1_-_An_introduction_to_ACN

redcactus, 2020-04-24:
Of course, I saw this topic. But there is no information about IMPORT :( . Also, data examples about length determinants contains errors. For this sample:
```
-- ASN.1 DEFINITION
TAP2File ::= SEQUENCE {
   nrCalls INTEGER,
   data SourceData
}
SourceData ::= SEQUENCE {
   operatorID IA5String,
   calls SEQUENCE (SIZE(1..100))OF Call
}
-- ACN DEFINITION
TAP2File [] {
   nrCalls [],
   data <nrCalls> [ ] -- nrCalls is passed as a parameter in SourceData
}
SourceData<INTEGER:nElements>
   -- nElements is a parameter used in encoding/decoding
   -- passed in from the levels above (in this case, TAP2File level)
{
   operatorID [],
   calls[size nElements] -- points to a parameter not a field
}
```
compiler will generate error.  As I understood from examples (in v4Tests
 folder), length determinant must not be defined in ASN, it allowed to be defined in ACN only.

maxime-esa, 2020-04-24:
You are right, some parts of the documentation seem a bit outdated. I fixed the section you spotted. The count field cannot indeed be in the ASN.1 grammar and must be introduced in ACN.

Regarding the IMPORT, a new section must be added.

redcactus, 2020-04-24:
Thanks in advance.
Is this a good idea to continue ask questions using this issue?

maxime-esa, 2020-04-24:
Of course, please do! That helps us spotting the places where documentation is lacking.