esa/asn1scc #201: CMake Module

multiplemonomials, 2020-06-07:
Hey guys!  I wrote [a CMake module](https://gist.github.com/multiplemonomials/b92e6d2bc6de228323039c694198f417) that allows you to easily use asn1scc in any CMake project.  I thought maybe you'd want to add it to this repo so that other people can find it. 

This module should make using asn.1 in CMake projects very simple.  After adding the cmake file someplace on your search path, you can just use:
```cmake
find_package(Asn1scc)
```

Then you can create asn.1 libraries like this:
```cmake
add_asn1_library(foo_serialization
    LIB_TYPE STATIC
    PROTOCOL foo.asn
    GEN_DIR ${CMAKE_CURRENT_SOURCE_DIR}/generated_c
    RULES uPER)
```
and the module will do all the heavy lifting.

ttsiodras, 2020-06-09:
Hi, @multiplemonomials - can you send this as a pull request for the "contrib" folder? You can also add a README in that folder describing how to use it.

multiplemonomials, 2020-06-10:
Will do!
Also, one other question: what's the difference between Asn1f2.exe vs Asn1f4.exe vs asn1.exe?  Which one should I be calling in the CMake module?

maxime-esa, 2020-06-10:
Asn1f2.exe is the old version of the compiler, you can ignore it. It should not be built anymore.
asn1.exe is just a renaming of Asn1f4.exe that is created by the build process. So you can use any of them.
For historic reasons we have been the single name asn1.exe in TASTE on Linux - where it's not common to have uppercases in executable filenames. But there is really no technical rationale behind.

Thanks for the work with cmake!

multiplemonomials, 2020-06-21:
Submitted!

closed 2020-06-21 by multiplemonomials