esa/asn1scc #202: --rename-policy 3 misses to rename some identifiers

joluxer, 2020-06-17:
I see the ``asn1scc`` compiler not to rename all occurences of a renamed identifiers, when using ``--rename-policy 3`` for C output. 

The exact situation happens randomly scattered throughout the generated C code, when using the determinant notation for ``CHOICE`` selection in ACN with ACN introduced selector field.

When falling back to ``--rename-policy 2`` this is not complained by the C compiler, perhaps because there need not all identifiers to be renamed.

Up to now, this all is tested with tag 4.2.1.3f.

usr3-1415, 2020-06-20:
I could not reproduce the error. I tried the test case ./v4Tests/test-cases/acn/09-CHOICE/010.asn1 with the --rename-policy 3 and no warning or error was emitted by the compiler.
Can you submit a minimal ASN.1/ACN grammar that reproduces the error?

joluxer, 2020-06-25:
I need some time to setup a test case from our current ASN.1 spec, but I'm working on that.

joluxer, 2020-07-01:
Up to now the only test case I could produce, was our productive ASN.1/ACN grammar, which produces the problems. 

If I want to supply you with that grammar, I'm obligated to scramble our identifiers. Our Grammar are 8 files (4 .asn, 4 .acn). Please, use the ZIP file attachment here.

The command, how I invoke the compiler is a one-liner which is normally in a shell script, that you can find in the ZIP file too. It's a docker call:

`docker run -it --rm -v $(pwd):$(pwd) -u "$(id -u):$(id -g)" -w $(pwd) joluxer/asn1scc:4.2.1.3f mono /opt/asn1scc/asn1.exe --c-lang --word-size 4 --fp-word-size 4 --acn-enc -o gen-test-scrambled --rename-policy 2 -atc "${@}" $(cat $(pwd)/processing-order.scrambled)`

[joluxer-asn1-tests.tar.gz](https://github.com/ttsiodras/asn1scc/files/4857559/joluxer-asn1-tests.tar.gz)

usr3-1415, 2020-07-13:
> I see the asn1scc compiler not to rename all occurences of a renamed identifiers, when using --rename-policy 3 for C output.
> The exact situation happens randomly scattered throughout the generated C [...]

It is very difficult to debug with a scrambled grammar, so I cannot find a case where an enumerant is not renamed. I tried to run multiple times using --rename-policy 3 and the asn1 compiler produced always is the same code (i.e. not random behavior). The code also compiles successfully and all automatic test cases pass.

Please provide a concrete case (even in your scrambled grammar) where an enumerant is not renamed (i.e. identify file and line).

>When falling back to --rename-policy 2 this is not complained by the C compiler, perhaps because there need not all identifiers to be renamed
Could not reproduce this behavior, When compiling with --rename-policy 2 the code compiles successfully and all automatic test cases pass.

closed 2020-07-13 by usr3-1415