esa/asn1scc #208: Load values from ASN.1 Value Notation

kliberty, 2020-12-08:
Is it possible to load values from a file that are written in the ASN.1 value notation? I'm working on an input deck for an application and am a fan of the ASN.1 syntax + validation.

With the schema
```
Prompt ::=  SEQUENCE {
    title       VisibleString    SIZE(1..100),
    message     SEQUENCE OF VisibleString    SIZE(1..7)
}
```
load a value from a text file into a native data structure
```
value Prompt ::= 
{
  title "My title",
  message 
  {
    "line 1",
    "line 2"
  }
}
```

maxime-esa, 2020-12-08:
Yes you can have the values in a separate file and have it compiled with the grammar.

In practice:

* Put your grammar in a file (e.g mygrammar.asn) with an ASN.1 Module named, eg. MyGrammar:
```
MyGrammar DEFINITIONS ::=
BEGIN
  Prompt ::= ...
END
```

* Use IA5String instead of VisibleString (which is not supported). ASN.1 comes with varieties of string types which are more or less similar. We chose to keep only IA5String for the sake of simplicity.

* Put bounds on your SEQUENCE OF, for instance
```
message SEQUENCE (SIZE (1..10)) OF IA5String (SIZE (1..7))
``` 

* The file with the values must be a valid ASN.1 grammar, so it must start with:
```
MyValues DEFINITIONS ::=
BEGIN
   IMPORTS Prompt FROM MyGrammar

   value  Prompt ::= ....

END
```

* Then compile the two files together:
```
asn1.exe -Ada mygrammar.asn myvalues.asn
```
or if you target C/C++:
```
asn1.exe -Ada mygrammar.asn myvalues.asn
```

And you will get the source code:
```
const Prompt value = {
    .title = "Hello",
    .message = {    .nCount = 2,    .arr = 
        {
            "line 1",
            "line 2"        
        }
    }
};
```

kliberty, 2020-12-09:
Thank you, I didn't actually know you could do that (still learning ASN.1). My original meaning was in a dynamic sense, so that users could provide input to an already built application.

maxime-esa, 2020-12-10:
Ah, this is different, this is what is called `GSER` (Generic String Encoding Rules), a standard textual encoding which we do not have support for at the moment (in the sense that we do not generate parsers and encoders for it in the target language).

closed 2021-01-15 by maxime-esa