esa/asn1scc #221: Docker hub's image is outdated

Nodraak, 2021-10-19:
Hi again!

*For an internal project at my company, I am investigating how to generate an ICD (html/pdf) from the ASN.1 definition file.*

The README.md specify that the easiest way is to use the docker image `ttsiodras/asn1scc`. On the [Docker hub](https://hub.docker.com/r/ttsiodras/asn1scc), this image is very old (5 years). One issue I noticed: it does not support the `--acn-enc` flag.

Could you update it? Alternatively, the README.md could be rewritten to instruct people to not use this option.

Thanks!

Nodraak, 2021-10-20:
Thanks a lot!

swthorn, 2023-09-19:
Hi,

going to bump this issue since the docker hub image does not look like it has been updated since this issue was created. Can it be automatically pushed after each commit?

Thanks!