esa/asn1scc #223: Can't compile gsm_map asn files to C sources

snowy-owll, 2022-01-11:
Hello!
I took asn files from [Wireshark](https://gitlab.com/wireshark/wireshark/-/tree/master/epan/dissectors/asn1/gsm_map) sources . Except for Ericsson.asn and Nokia.asn files. I also added the necessary from [ros](https://gitlab.com/wireshark/wireshark/-/tree/master/epan/dissectors/asn1/ros).
To compile I use the following commands:
```
$gsm_map = (Get-ChildItem d:\gsm_map -File).FullName
asn1scc.exe -c -uPER -XER -o d:\gsm_map_out $gsm_map
```
After execution, I get the following error:
`D:\gsm_map\GSMMAP.asn:747:2: error: missing DOT at 'SendParametersArg'`
Is it possible to do something about this and compile these asn files?
All hope for your compiler :)

OS: Windows 10
asn1scc.exe compiled from master branch.

maxime-esa, 2022-01-11:
This module is using Object Information Classes, which is an ASN.1 construct that our compiler currently does not support:

[image]

snowy-owll, 2022-01-11:
Thanks for the quick response!
That is, there are only paid compilers left or somehow pulling decoders from the Wireshark source codes?

maxime-esa, 2022-01-11:
I am not sure about that, as I am not too familiar with all features of other open-source compilers.  Did you check them out already?

This construct (Object Information Classes) is mostly related to the description of interfaces (like function prototypes) and not directly to the description of messages (which is done via ASN.1 types). This is why we do not support it, as it goes a bit out of the scope of the language and somehow overlaps with other modelling techniques that we use.

An option is to remove these constructs from the input grammars and keep only the ASN.1 types - after all, they are the part that matters to get binary encoders and decoders.

snowy-owll, 2022-01-11:
I have tried the [asn1c](https://github.com/vlm/asn1c) compiler. In questions on the github, I met that he also does not support Object Information Classes. And it also gives an error when trying to compile. I did not find any other free compilers in C/C++ with uPER support.
Thanks for the advice! I will try to understand and remove these constructs.

closed 2022-01-11 by snowy-owll