esa/asn1scc #228: Some components need to be defined only in ACN file so as to be used as determinants, but then no type definition is generated for them

alk-i, 2022-02-17:
First time I try to report something so I hope I do it correctly.

I am not sure if there is a reason behind this, but in case a field is defined only in ACN file then the generated encapsulating struct does not contain any member with such name.

In the following code example, if I define ServiceTypeID in the ASN file then I get the following error:

_error: ASN.1 fields cannot act as encoding determinants. Remove field 'serviceTypeID' from the ASN.1 grammar and introduce it in the ACN grammar_

If I do not define it in the ASN file however then no struct member is generated for it and as a result I cannot access its value in my C code.
  

```
secondaryHeader [] {   
    pusVersion [],   
    ackFlags [],   
    serviceTypeID    T-uint8 []   
},    
telecommand <secondaryHeader.serviceTypeID> []    
```

Is this the expected behavior or is it indeed an issue of the tool?

maxime-esa, 2022-02-17:
Yes, it is the expected behaviour and there should be no need to access this field in your C code because the information is implicitly carried by the CHOICE element (so you have access to it already).

In your example I assume that `telecommand` is a `CHOICE` type, right? The service type ID is the choice determinant only at binary level. At code level (in C) you have a data structure that contains a union _and a `.kind` field_  which is the determinant you can use.

The `.kind` field is what you have to set when you create the telecommand prior to encoding, and what you can read after decoding the telecommand. How it is actually encoded at binary level (the `serviceTypeID` ACN-added field) is only useful by the tools to ensure that the packet respects the PUS format.

I hope that's clear?

alk-i, 2022-02-17:
Thank you very much for you quick response!

I understand your logic, indeed `telecommand` is a `CHOICE` type and I have access to it through the `.kind` field. 
For my C code architecture however it would be more convenient if I had access as well to the `serviceTypeID` value itself. 
The reason is the following, the code I provided is actually simplified, the determinants are two (`serviceTypeID` and `messageSubTypeID`) and it would be convenient to group together in conditions telecommands for which the service type is common, otherwise I would need to define separate mapping arrays or functions based on the `.kind` enumeration.

This is a minor issue though and I can implement workarounds, I wanted to confirm that this is indeed the expected behavior of the tool and if in the future the determinant values are also accessible in the generated structures I think it would be better.

Once more, thanks for responding quickly and for your explanation!

closed 2022-02-17 by alk-i

maxime-esa, 2022-02-17:
I think that to do that, you better define two levels of CHOICE as explained in this tutorial:
http://taste.tuxfamily.org/wiki/index.php?title=Technical_topic:_Hints_to_model_complex_packet_encodings_with_ASN.1_and_ACN#Create_the_ASN.1_Structure

Based on this structure you can easily switch-case to dispatch all TCs of a given service type with a single test of the first level of `.kind` field.

Check also the ACN structure using this baseline.