esa/asn1scc #229: Ada keywords are not allowed as field names, even when generating only C code

ar-nelson, 2022-02-24:
When generating C source code with `-c`, ASN1SCC fails if any Ada keywords are used as identifiers in the ASN.1 definitions. This prevents the use of common words like `type`, even if there is no reason to forbid these words when not generating Ada code.

Ideally, ASN1SCC would only check for C keywords when generating C code and Ada keywords when generating Ada code.

maxime-esa, 2022-02-24:
The solution is to use the `--field-prefix` (or just `-fp`) flag with value set to `AUTO`:

```
$ asn1scc -fp AUTO -c myGrammar.asn
```

In that case the field `type` will be accepted as is.

 In combination with `-Ada` however it will be renamed:
```
$ asn1scc -fp AUTO -Ada myGrammar.asn
[INFO] Renamed field "type" in type "PUS-C.TC" to "TC_type" ("type" is a Ada keyword)
```

closed 2022-03-31 by maxime-esa

ar-nelson, 2022-04-13:
Thank you, this fixes the issue. (Sorry for waiting so long to respond.) My only suggestion is that it would be helpful to add a hint about this feature to the error message when a keyword is encountered and `-fp` is not set.

maxime-esa, 2022-04-14:
Agreed, we will. Thanks.