esa/asn1scc #230: Performance issues in BIT STRING types encoding and decoding

bonizzifbk, 2022-02-25:
When dealing with large BIT STRING types, there is a critical lack in performance of the encoding and decoding routines that can break the real time requirements of the application.

Currently, the encoding of large BIT STRING type is a costly process, because it includes copying large quantities of data bit per bit. This is unavoidable when the type is not byte aligned (general case), but in the case it is (for example, because the user forced this with ACN) it is possible and more efficient to copy the data on a per byte or word basis.

In the attached example, I substituted _BitStream_AppendBits()_ with a stdlib's _memcpy_ call, and achieved on my test machine a speedup of 2 magnitude orders.

The proposed change is safe to make, because if the _to-encode-attribute_ is byte aligned, we have the guarantee that inside the while loop of the encoding function the chunks of data to copy (of size `nCurBlockSize1`) are all byte aligned. The control values inserted inside the buffer (`C1` to `C4`) are all one byte sized and so they do not break the alignment.

Optionally, an if guard could be added to allow the fast copy whenever the current bit offset is `0`, otherwise to proceed with the _BitStream_AppendBits()_ approach.

The same principle can be ported to the decoding process.

[demo_project.zip](https://github.com/ttsiodras/asn1scc/files/8140505/demo_project.zip)

roboogle, 2022-02-25:
In addition, such performance improvement may be implemented also for encoding/decoding functions in other types, like for example octet strings.

usr3-1415, 2022-03-30:
Noted it. I will look into it.

maxime-esa, 2022-09-29:
@roboogle @bonizzifbk 
The issue was fixed in the last release (https://github.com/ttsiodras/asn1scc/releases) and integrated in TASTE

closed 2022-10-01 by maxime-esa