esa/asn1scc #231: Missing C function when SIZE constraint is duplicated on a SEQUENCE member

Glamhoth, 2022-03-14:
Let's consider the following ASN.1 file:
```
MYMODULE DEFINITIONS ::= BEGIN

MyArray ::= SEQUENCE (SIZE(20)) OF INTEGER (1..20)

MySeq ::= SEQUENCE
{
    data MyArray(SIZE(20))
}

END
```

C file that is generated from this file, by running
```
$ asn1scc -c -ACN mymodule.asn
```  
tries to call `MySeq_data_elem_Initialize` which was not created
[image]

usr3-1415, 2022-03-30:
Thanks for reporting this, I will look into it.

maxime-esa, 2022-11-15:
The bug is still present in the latest version (`4.4.0.2`).
There is a workaround however: when using the new `-ig` flag, the generated code is correct (no symbol is missing):
```
const MySeq_data MySeq_data_constant = {.arr  = {[0 ... 20-1] = 1UL }};
const MySeq MySeq_constant = {.data = MySeq_data_constant};

void MySeq_data_Initialize(MySeq_data* pVal)
{
	(void)pVal;
	(*(pVal)) = MySeq_data_constant;
}
```

This flag makes the compiler generate constants to initialize the data structures, instead of code (with loops for the arrays).

usr3-1415, 2023-01-06:
fixed

closed 2023-01-06 by usr3-1415