esa/asn1scc #235: Compiler crashes on mutually exclusive ENUMERATED constraints

Glamhoth, 2022-05-19:
Let's consider the following ASN.1 file:

```
CrashingEnum DEFINITIONS ::= BEGIN

MyEnum ::= ENUMERATED
{
    value1,
    value2,
    value3
}

MyConstrainedEnum ::= MyEnum (value1) (value2)

END
```

Trying to compile that file crashes the compiler with error:
```
The input list was empty.
   at Microsoft.FSharp.Collections.FSharpList`1.get_Head() in D:\a\_work\1\s\src\fsharp\FSharp.Core\prim-types.fs:line 3742
```

usr3-1415, 2022-05-19:
Noted. This is a bug and will be fixed.
However, please note that the new type doesn't allow any values.
The first constraint   MyEnum (value1) limits the enum values to value1 and the second one to none (since value2 is not in the set of allows values any more).

If your intention was to limit, MyConstrainedEnum  to value1 and value2 then try this
MyConstrainedEnum ::= MyEnum (value1 | value2)

Glamhoth, 2022-05-19:
Yes, you are right, such type doesn't allow for any values and an error should be reported. My intention was to test what XMLs would `asn1scc` output on some strange cases. ;]

usr3-1415, 2022-05-21:
Fixed

closed 2022-05-21 by usr3-1415