esa/asn1scc #240: Named bit string positions with UPER

aosterhage, 2022-08-05:
I'm utilizing asn1scc with UPER encodings in C. I can't find the appropriate information in the ITU-T X.691 document about encoding bit strings but examining several different ASN.1 compilers it seems that bit strings are flipped in UPER from how they are defined. E.g. given this definition of a bit string:
`
BitString BIT STRING ::= {
  first(0),
  second(1),
  third(2)
} (SIZE(3))
`
this data:
`
someBitString BitString ::= { first, second, third }
`
will be encoded to:
`
0x e0
0b 1110 0000
`

My confusion comes from the included bit string names provided by asn1scc (per #196). It will provide something similar to:
`
#define BitString_first 0x1  /*(1 << 0)*/
#define BitString_second 0x2  /*(1 << 1)*/
#define BitString_third 0x4  /*(1 << 2)*/
`
which seemingly are unusable for constructing or examining any BitString types because the C library doesn't seem to do any conversions. Is this expected or do I have a misunderstanding somewhere?

maxime-esa, 2022-08-05:
@usr3-1415 will correct me if I am wrong but I think you are right and this is what I reported in this comment : https://github.com/ttsiodras/asn1scc/issues/196#issuecomment-581307122

aosterhage, 2022-08-05:
Thanks for the link, I missed that comment when going through the thread

usr3-1415, 2022-09-04:
A new set of functions is now generated that help programmer to set the bit string.
In this case, these functions will be called BitString_set_first(), BitString_set_second() etc.
There functions are not called during the encoding or decoding process.

closed 2022-09-04 by usr3-1415