esa/asn1scc #249: 3GPP 38331 h20 standard issue, error: Cyclic dependencies detected

Scherholz, 2022-10-14:
Hi, 

I am trying to use asn1scc to compile the 3GPP standard 38331 h20 ASN1 into C , and I got the following error:

38331-h20.asn:54:0: error: Cyclic dependencies detected : Type assignment 'NR-RRC-Definitions.DL-DCCH-Message' depends on : 'NR-RRC-Definitions.DL-DCCH-MessageType' ... 
[asn1sccErrorLog.txt](https://github.com/ttsiodras/asn1scc/files/9788551/asn1sccErrorLog.txt)

https://www.3gpp.org/ftp/tsg_ran/WG2_RL2/Specifications/202209_final_specs_after_RAN_97

I used a python script to parse the file 38331-h20.docx asn sections. 

[38331-h20asn.docx](https://github.com/ttsiodras/asn1scc/files/9788580/38331-h20asn.docx)

Could this be an issue with ASN1SCC , the specification or the way I parsed the file? 

Best Regards,
Luis Dias

maxime-esa, 2022-10-14:
Hi Luis,
Is your grammar recursive? That would be causing this error.
For example if you have such a grammar:

```
D DEFINITIONS ::=
BEGIN
  Foo ::= SEQUENCE {
    a Bar
  }

  Bar ::= SEQUENCE {
    x Foo
  }
END
```

You will get a similar error:
```
error: Cyclic dependencies detected : Type assignment 'D.Foo' depends on : 'D.Bar' and Type assignment 'D.Bar' depends on : 'D.Foo'
```

ASN1SCC generates code with static memory allocation. A recursive type by definition may take unbounded space, and therefore this is currently not supported.

maxime-esa, 2022-11-15:
@Scherholz I close the issue, please re-open if you think I missed something in my answer

closed 2022-11-15 by maxime-esa