esa/asn1scc #263: Recursive _IsConstraintValid()

escmmo, 2023-04-15:
asn1scc version 4.5.0.4
The compiler produces an _IsContraintValid() function that calls itself unconditionally:

```
flag T_STRUCT_member_IsConstraintValid(const T_STRUCT_member* pVal, int* pErrCode)
{
    flag ret = TRUE;
    ret = T_STRUCT_member_IsConstraintValid(pVal, pErrCode);
    if (ret) {
        ret = ((*(pVal)) == 100UL);
        *pErrCode = ret ? 0 :  ERR_T_STRUCT_MEMBER_2_2; 
    }   /*COVERAGE_IGNORE*/

        return ret;
}
```

The .asn definition that generates this is:

```
T-uint32 ::= INTEGER(0..4294967295)

T-SOME-TYPE ::= T-uint32

T-STRUCT ::= SEQUENCE
{
	member	T-SOME-TYPE (100)
}
```

The key points which trigger this behavior are:
- defining one type on top of another type (`T-SOME-TYPE` on top of `T-uint32`
- having the SEQUENCE `member` of that type with a specified value (100).

If `member` is defined as `T-uint32` instead, no recursive function is created.

maxime-esa, 2023-04-20:
Reproduced. We are looking into it, thanks for reporting.

usr3-1415, 2023-05-06:
Issue Fixed.
Thanks for reporting the issue.

closed 2023-05-06 by usr3-1415