esa/asn1scc #269: Test fails

ecshsun, 2023-06-29:
Executed tests on ARM64, macOS Ventura 13.3.1 (a) (22E772610a)
**Note**: using .NET 7.0.305 (README says 6.0, but that does not work).
```
c ACN /Users/henrik/dev/tools/asn1scc/v4Tests/test-cases/acn/09-CHOICE/012.asn1
09-CHOICE/012.asn1 MyPDU[]
Asn.1 compiler failed
Asn.1 compiler error is: /Users/henrik/dev/tools/asn1scc/v4Tests/tmp_c/sample1.acn:3:0: warning: The ACN module 'TEST-CASE' contains type assignments that do not exist in the corresponding ASN.1 module. For example, assignments 'MyPDU' are not defined in the ASN.1 module.

make: *** [cTests] Error 1
```

usr3-1415, 2023-10-07:
Sorry for the delayed response. I am currently unable to reproduce the issue; the test case 09-CHOICE/012.asn1 runs successfully on my end. By reviewing the file's history, I noticed there was a version where the ACN file incorrectly referenced a non-existing ASN.1 type assignment named "MyPDU." However, this seems to have been rectified. Could you please confirm if the issue has been resolved for you? If so, may we proceed to close this ticket?

maxime-esa, 2023-10-08:
Confirmed, it was fixed in this commit (as part of the integration of the new Scala backend):
```
003d47bb1ef744d6c7a1559a29aa70bd357d8f1a
Author: Ivo Nussbaumer <ivo.nussbaumer@ateleris.ch>
Date:   Tue Jul 11 16:53:12 2023 +0200
```

closed 2023-10-08 by maxime-esa