esa/asn1scc #271: Outdated docker hub image

swthorn, 2023-09-19:
[image]

It doesn't look like the docker hub image has been updated since #221 was closed. Can it be automatically pushed after each commit?

Thanks!

usr3-1415, 2023-10-07:
I've created a GitHub Action with two jobs to automatically build and push the Docker image after each commit. The build job is triggered automatically after a commit, but only in the master branch. The push image job is triggered manually via GitHub's Action page. However, the push image job requires the `secrets.DOCKERHUB_USERNAME` and `secrets.DOCKERHUB_PASSWORD` secrets to be set. Unfortunately, I am not the owner of this repository and cannot set these secrets myself.

@maxime-esa, would it be possible for you to set these variables so that the GitHub Action can push the Docker image to Docker Hub? Thank you!

maxime-esa, 2023-10-08:
@usr3-1415 No, I am also not the owner. @ttsiodras ?

ttsiodras, 2023-10-08:
Since I left the Agency more than a year ago, I think the best course of action is to transfer ownership to you @maxime-esa , no?  You'll have to update the links in the TASTE repos to yourself, of course.

Shall I?

maxime-esa, 2023-10-08:
Sure please do @ttsiodras 
Thanks!

ttsiodras, 2023-10-08:
"maxime-esa/asn1scc already exists" - I guess you have a fork you need to remove first.

maxime-esa, 2023-10-08:
@ttsiodras I have deleted the fork I had

ttsiodras, 2023-10-09:
...and I have initiated the transfer. Check your email, Github must have sent you a confirmation - you need to accept the transfer for it to go through.

maxime-esa, 2023-10-09:
Thanks... Done!  

Back to the original problem now... @usr3-1415 I found the area where I can create "secrets" in the Github repo. So I can indeed create `secrets.DOCKERHUB_USERNAME` and `secrets.DOCKERHUB_PASSWORD`. Can you let me know (privately) what values I should use?

maxime-esa, 2023-11-17:
As stated in #280 we won't maintain the docker container as there are pre-built releases that work out of the box.

closed 2023-11-17 by maxime-esa