esa/asn1scc #274: Empty ASN1 SEQUENCE produces incorrect C code

deiruch, 2023-09-26:
https://github.com/n7space/asn1-pusc-lib/blob/master/service-12/PUS-12-9.asn1 contains the following borderline-incorrect ASN1 specification:
```ASN1
PUS-12-9 DEFINITIONS ::= BEGIN

TM-12-9-ParameterMonitoringDefinitionReport ::= SEQUENCE
{
}

END
```

# Expected behaviour

- ASN1SCC does not output the struct at all
- ASN1SCC emits an error message and stops code generation

# Actual behaviour
ASN1SCC produces the following, non-compiling C code from the ASN1 specification:
```C
typedef struct {

} TTM_12_9_ParameterMonitoringDefinitionReport;
```

maxime-esa, 2023-09-27:
I don't understand - what error do you have ? This seems to be correct C code.
I tried to compile it and got no errors - even no warnings:

[image]

deiruch, 2023-09-27:
Technically, it's not valid C code. Quoting the C99 standard (via https://stackoverflow.com/a/15416194/742404):
> — A structure type describes a sequentially allocated nonempty set of member objects (and, in certain circumstances, an incomplete array), each of which has an optionally specified name and possibly distinct type.

MSVC will not compile the code, and bail with
> error C2016: C requires that a struct or union has at least one member

maxime-esa, 2023-09-27:
I see, and I assume that the same happens with zero-length arrays?

In the following grammar:
```
 Zero ::= OCTET STRING (SIZE (0))

 Zero2 ::= IA5String (SIZE (0))

 Zero3 ::= SEQUENCE (SIZE (0)) OF Zero
```

The compiler will generate

```
typedef struct {

    byte arr[0];
} Zero;

typedef char Zero2[1];  // This one is correct because an extra byte is added for the string null termination

typedef struct {
    Zero arr[0];
} Zero3;
```  
but C99 also disallows zero-size arrays so I am guessing that MSVC also fails? (gcc does not complain)

Since this is valid ASN.1, we should update the compiler to produce:

* a dummy member in the struct to cover empty `SEQUENCE`
* arrays of size 1 when the ASN.1 grammar says 0

@usr3-1415 What do you think ?

usr3-1415, 2023-10-02:
I agree with Maxime's proposal and will incorporate it through two optional command line arguments: `--handle-empty-sequence` and `--handle-zero-arrays`.

- When `--handle-empty-sequence` is used, a "dummy" int member will be added to empty sequences to address the issue.
- Similarly, when `--handle-zero-arrays` is present, the compiler will generate arrays of size 1 whenever the ASN.1 grammar specifies a size of 0, ensuring compatibility with C99 and other compilers that disallow zero-size arrays.

usr3-1415, 2023-10-04:
Closing this issue as resolved with the introduction of the `--handle-empty-sequence` flag, which when set, adds a dummy integer field to empty ASN.1 SEQUENCE structures to ensure compliant C code generation. The `--handle-zero-arrays` flag was not added as zero-sized arrays compile successfully with both GCC and MSVC. The option to use C99's flexible array members was explored but caused issues in GCC, hence no changes were made to the type definitions regarding zero-sized arrays.

closed 2023-10-04 by usr3-1415