esa/asn1scc #280: docker container command does not work for me.

HelenHZhang, 2023-11-09:
I tried to follow the instruction

> docker pull maxime-esa/asn1scc
> docker run -it maxime-esa/asn1scc

I got the following error in the first command:
Error response from daemon: manifest for dockerhub.cisco.com/docker.io/maxime-esa/asn1scc:latest not found: manifest unknown: The named manifest is not known to the registry.

Please advise me. Thanks

Sedictious, 2023-11-14:
Hello, I am not one of the maintainers but I think it is related to #271 

For me the older image ttsiodras/asn1scc works (granted that it is 2 years old) or alternatively build the image yourself (you should be able to get the necessary commands from the [config](https://github.com/maxime-esa/asn1scc/blob/master/.circleci/config.yml))

maxime-esa, 2023-11-17:
We are not maintaining the `docker` option so this will be removed from the `README.md` file in the next version. 

Please use the available Releases (on Github) to get a working binary. At the moment only the Linux releases are up to date but if you need a Windows binary I can build it and upload it as well.

closed 2023-11-17 by maxime-esa